eclipse-milo/milo #5: Always build source attachments

ctron, 2016-07-05:
Currently the build only generates javadoc and source attachments when the "release" profile is active.

I would suggest to always generate the source and javadoc attachments so that it is possible to work with the source code in the case of -SNAPSHOT versions.

kevinherron, 2016-07-05:
Hmm. Let me see how much longer it takes to build when including javadocs.

I think, if nothing else, I should deploy snapshots with release profile active.

ctron, 2016-07-05:
Would you just could add some "withDocs" profile. Which does the trick.

ctron, 2016-07-05:
Thanks :+1: So this gets to deployed to the maven repository then, even for -SNAPSHOT builds?

kevinherron, 2016-07-05:
You have to activate this profile when you package or deploy, but yes I'll deploy SNAPSHOT builds with javadoc and source now.

locally: `mvn clean install -P javadoc-and-source`
deploying: `mvn clean deploy -P javadoc-and-source` 

I've deployed a version to sonatype that includes javadoc and source.

ctron, 2016-07-05:
Thanks a lot!

closed 2016-07-05 by kevinherron