eclipse-milo/milo #6: opc client code does not always reconnect successfully

sandeep-foghorn, 2016-07-05:
Sometimes, the opc client code fails to connect.

Here's successful connection logs

```
2016-07-01 21:05:07 DEBUG UaTcpClientAcknowledgeHandler:150 - Sent Hello message on channel=[id: 0x4f988c56, /172.18.0.5:44544 => /10.83.202.40:49320].
2016-07-01 21:05:07 DEBUG UaTcpClientAcknowledgeHandler:208 - Received Acknowledge message on channel=[id: 0x4f988c56, /172.18.0.5:44544 => /10.83.202.40:49320].
2016-07-01 21:05:07 DEBUG UaTcpClientMessageHandler:153 - OpenSecureChannel timeout scheduled for +5s
2016-07-01 21:05:07 DEBUG UaTcpClientMessageHandler:220 - Sent OpenSecureChannelRequest (Issue, id=0, currentToken=-1, previousToken=-1).
2016-07-01 21:05:07 DEBUG UaTcpClientMessageHandler:378 - OpenSecureChannel timeout canceled
2016-07-01 21:05:07 DEBUG UaTcpClientMessageHandler:422 - Received OpenSecureChannelResponse.
2016-07-01 21:05:07 DEBUG UaTcpClientMessageHandler:500 - SecureChannel id=118612, currentTokenId=1, previousTokenId=-1, lifetime=3600000ms, createdAt=DateTime{utcTime=131118806733609274, javaDate=Fri Jul 01 21:04:33 UTC 2016}
2016-07-01 21:05:07 DEBUG UaTcpClientMessageHandler:128 - 0 message(s) queued before handshake completed; sending now.
2016-07-01 21:05:07 DEBUG ClientChannelManager:178 - Channel bootstrap succeeded: localAddress=/172.18.0.5:44544, remoteAddress=/10.83.202.40:49320
2016-07-01 21:05:07 DEBUG ClientSessionManager:431 - Sending (re)ActivateSessionRequest, secureChannelId=118612, channel=[id: 0x4f988c56, /172.18.0.5:44544 => /10.83.202.40:49320]...
2016-07-01 21:05:07 DEBUG ClientChannelManager:219 - Reconnect succeeded, channelId=118612
2016-07-01 21:05:07 DEBUG ClientSessionManager:448 - Session reactivated: NodeId{ns=1, id=143dd10c-2ac8-4ae0-9361-7e4aaf0414f6}
2016-07-01 21:05:07 DEBUG OpcUaSubscriptionManager:339 - Sending PublishRequest, requestHandle=2573, acknowledgements=[id=1049/seq=2566]
2016-07-01 21:05:07 DEBUG OpcUaSubscriptionManager:339 - Sending PublishRequest, requestHandle=2574, acknowledgements=[]
2016-07-01 21:05:08 DEBUG OpcUaSubscriptionManager:349 - Received PublishResponse, sequenceNumber=2567
2016-07-01 21:05:08 DEBUG OpcUaSubscriptionManager:339 - Sending PublishRequest, requestHandle=2575, acknowledgements=[]
2016-07-01 21:05:08 DEBUG OpcUaSubscriptionManager:377 - onPublishComplete() response for subscriptionId=1049
2016-07-01 21:05:08 DEBUG OpcUaSubscriptionManager:427 - [id=1049] PublishResponse sequence=2567, available sequences=[id=1049/seq=2567]
2016-07-01 21:05:08 DEBUG OpcUaSubscriptionManager:495 - onNotificationMessage(), subscriptionId=1049, sequenceNumber=2567, publishTime=DateTime{utcTime=131118806742501290, javaDate=Fri Jul 01 21:04:34 UTC 2016}
2016-07-01 21:05:08 DEBUG OpcUaSubscriptionManager:510 - Received 6 MonitoredItemNotifications
```

kevinherron, 2016-07-05:
What server is this?

sandeep-foghorn, 2016-07-05:
Kepware

kevinherron, 2016-07-05:
This may be an outstanding bug that we've reported to Kepware. We're running into an issue that might be the same one at work. If you can turn off security for the connection and get me a Wireshark capture I can confirm.

sandeep-foghorn, 2016-07-05:
Can you please provide more details on the Kepware issue ? And, thanks for helping out !

kevinherron, 2016-07-05:
Basically, what happens is that Kepware simply stops responding to requests, including PublishRequests, after a successful reconnect happens. I don't really have more details on why. They've issued a patch to some customers to try and address the issue, but I haven't heard back from anybody on whether it works definitively. The patch we got was for Kepware v5.20.

sandeep-foghorn, 2016-07-05:
Ok. Is there a way for me to get the patch ? Or do we need to contact Kepware for that ?

kevinherron, 2016-07-05:
I think you may need to contact Kepware. If they aren't sure what you're talking about, reference the OPC-UA issue between Inductive Automation's Ignition SCADA and KEPServer.

sandeep-foghorn, 2016-07-05:
Let me figure out how we can do that.

kevinherron, 2016-07-06:
I'm going to close this issue for now. We can continue discussing this on the mailing list or by direct email if necessary.

closed 2016-07-06 by kevinherron