eclipse-milo/milo #11: Server: Unable to write to node of type "Variant"

ctron, 2016-07-07:
I just tried writing a `String` to a Variable Node of Type `Variant`, which fails with "Bad Type".

Debugging into "AttributeWriter.validateDataType" it comes up with a `DataValue`, which gets unboxed to `Variant` and then to `String`. But then the check tries to assign a String to the target Variant, which fails of course.

I would expect that it is possible to write "anything" to a node of type `Variant`.

kevinherron, 2016-07-07:
I'll have to look into whether that should be possible or not. You can write "anything" do a node of type `Variant` - you just have to encapsulate that anything in a `Variant` first.

ctron, 2016-07-07:
That is what should happen .. at least I think it is.

So on the client side I am doing something like `write ( new DataValue ( new Variant ( "String" ) ) )`

On the server side it pops up like in the following screenshot:

[image]

So the original value is the DataValue, from there the Variant and then the String.

Or do you mean I should put a Variant inside the Variant?

kevinherron, 2016-07-07:
You can't put a Variant in a Variant - I think you're right about this being a bug.

ctron, 2016-07-07:
Yes I just saw that:

```java
checkArgument(clazzIsArray || !Variant.class.equals(componentClazz), "Variant cannot contain Variant");
```

ctron, 2016-07-07:
Since it currently does block me, I can work on a fix if you like.

kevinherron, 2016-07-07:
Sure, in that block you've breakpointed in, you'll need to add a check to see if expected is `Variant`, and if so, allow the write to proceed.

If you could start a unit test for AttributeWriter that would be good too - you could just test this case for now.

ctron, 2016-07-07:
Alright then ... yes, you are right. A unit test seems to be a good way to start.

closed 2016-07-07 by kevinherron