eclipse-milo/milo #9: Possible even loss on monitored item

ctron, 2016-07-07:
I have to admit, I am not 100% sure if I am correct on this one.

The following code is in the class `OpcUaSubscriptionManager`

```java
void onValueArrived(DataValue value) {
   BiConsumer<UaMonitoredItem, DataValue> c = valueConsumer;
   if (c != null) c.accept(this, value);
}
```

And items are subscribed using the following pattern (from `OpcUaClientIT`):

```java
 List<UaMonitoredItem> items = subscription.createMonitoredItems(TimestampsToReturn.Both, newArrayList(request)).get();

// do something with the value updates
UaMonitoredItem item = items.get(0);

CompletableFuture<DataValue> f = new CompletableFuture<>();
item.setValueConsumer(f::complete);
```

Now this seems to me like a non-atomic two step approach. Between the call to "create" and `setValueConsumer` the `valueConsumer` would be `null` and thus `onValueArrived` would not fire.

It seems to me that this may cause the first event to be lost.

kevinherron, 2016-07-07:
This race condition is discussed here: https://github.com/digitalpetri/ua-client-sdk/issues/8

To fix it beyond using the MonitoringMode to control when values get published going to require some fundamental changes to the way the client works or to the API for creating items. I have not really decided what to do about it yet...

kevinherron, 2016-07-07:
The more I look at this the more I think the only good solution is to change the API. Anything else will result in the API being non-obvious to use.

Something like this:
```java
createMonitoredItems(TimestampsToReturn timestampsToReturn,
                     List<MonitoredItemCreateRequest> itemsToCreate,
                     Consumer<UaMonitoredItem> itemCreationCallback);
```

Where the callback is your opportunity to set value and event consumers on the item after it is created, with the guarantee by the implementation that it will prevent values from arriving before your consumers are set so long as you do it during the callback.

ctron, 2016-07-07:
Wouldn't it a possibility to pass in the consumer which you later set anyway?

kevinherron, 2016-07-07:
You have to allow for them to set a Consumer/BiConsumer of both DataValue and Variant (for value and event subscriptions). This would require 4 different overloaded calls and you would not be able to create value and event items in the same call.

kevinherron, 2016-07-08:
Please have a look at the proposed fix in PR https://github.com/eclipse/milo/pull/13

closed 2016-07-13 by kevinherron