eclipse-milo/milo #15: What is the difference between setValueConsumer and setEventConsumer

tosuns, 2016-07-08:
I'm trying to understand what the difference is between the two mentioned method calls.

if I look at the consumer declaration of the `setEventConsumer`, it seems like there is no designated event structure, because the consumer will get an `Variant[]` array.

I guess the `Variant[]` array contains attribute values of the `UaMonitoredItem`, like sourcetimestamp, servertimestamp, writemask an so on ?

And the `setValueConsumer` does only listen to value changes of the `UaMonitoredItem` ?

kevinherron, 2016-07-08:
Events are a special kind of monitored item; the result of subscribing to the Event Notifier property of a Node in which the value of the property is non-zero. When you subscribe to events your provide a filter which can narrow down not only which types of events you receive but which fields of that event type your interested in. The `Variant[]` your event consumer expect will correspond to those fields.

The server doesn't support events yet, so there's no example test case to explain it better, but this gist I put together a while back on how to subscribe to events in the client may clarify: https://gist.github.com/kevinherron/0143b38cc93e19266156

kevinherron, 2016-07-08:
Need better javadoc on `setEventConsumer` methods.

tosuns, 2016-07-08:
I have a question regarding the `QualifiedName` in your example. Is the `QualifiedName` relative to the `readValueId` or is it an absolute ?

Another question pops up: Do I understand it correctly that every node can be defined as an Event supplier/notifier? As far as I can see, there is an attribute identifier `EventNotifier`, which indicates that is the case !?

kevinherron, 2016-07-08:
The QualifiedName is determined by which event "fields" you're interested in. In the address space, all of the fields of an event will be property nodes of the event object.

All ObjectNodes have an EventNotifier attribute, but they can only be subscribed to for events if the attribute is set to 1. It's typical for a server to use the Server object as its event notifier, but there's nothing requiring it.

OPC-UA specification part 3, 4, and 5 would all help clarify this bit. Unfortunately OPC-UA is so complex that it's probably not possible to understand it without the aid of the specifications.

closed 2016-07-11 by kevinherron