eclipse-milo/milo #16: How to determine the namespace of an given identifier

tosuns, 2016-07-08:
I'm trying to figure out, whether there is a convenient way to determine the namespace or better the `NodeId` of an data point (`DataValue`) if only the associated identifier is provided ?

For my use case those opc ua namespaces are more annoying instead of convenient :/ 

Another question: Is it actually specified that `DataValue` should not have the associated identifier/nodeId ?

kevinherron, 2016-07-08:
I'm not sure what you mean by "only the associated identifier" - are you talking about the client handle you assign a monitored item when creating it, or something else?

tosuns, 2016-07-08:
If I want to read the value of a node I have to provide a `NodeId` first. One of the constructors is `NodeId(int index, String identifier)`.
In my client code I don’t want to mess with namespace indices. I only want to provide the identifier and the opc ua client should somehow figure out to which namespace the identifier belongs to.

I want the opc ua client to be one of possible client implementations, that could be used. That is why I try to create a client facade/service to possible servers, like opcua or beckhoff ads, bachmann and so on. Not all of them have a concept of namespaces, but alot of them have the concept of identifiers/symbols.

kevinherron, 2016-07-08:
You always need the NodeId.

I think the only alternative would be using the TranslateBrowsePathsToNodeIds service, but you still need to be storing the whatever it is in a format that you could turn into a `BrowsePath` for the translate service call.

ctron, 2016-07-11:
I actually stumbled over the same issue. In order to access a variable you also need the NodeId, and therefore the index. However to me it seems (as far as I look into the server side code) that the index is not static as the URI is. So there is some sort of mapping on the server side from URI -> Index.

So assuming in the client that the index never changes seems to be wrong to me, and going with the URI is the better approach. What I did is that I do make a lookup on the server table:

```java
CompletableFuture<DataValue> future = this.client.readValue(0, TimestampsToReturn.Neither,
   Identifiers.Server_NamespaceArray);

   return future.thenApply(value -> {
      final Object rawValue = value.getValue().getValue();

      if (rawValue instanceof String[]) {
         final String[] namespaces = (String[]) rawValue;
         for (int i = 0; i < namespaces.length; i++) {
         if (namespaces[i].equals(namespaceUri)) { return Unsigned.ushort(i); }
      }
   }
return null;
});
```

I am not sure if this is the proper way and I also could imagine that the lookup result could be cached on the client side.

kevinherron, 2016-07-11:
So what @ctron is suggesting is the most correct approach if you're going to be storing NodeIds. In practice I have never actually seen the need for this, as all the servers we deal with tend to have fixed namespaces 0, 1, and 2. (0 will never change. 1 is the namespace URI of the server; it shouldn't change. 2+ may change depending how the server uses namespaces).

ctron, 2016-07-11:
So what is the correct way to extend the #1 namespace?

kevinherron, 2016-07-11:
Right now the SDK registers a pretty sparse Namespace implementation called `VendorNamespace` as namespace 1. You could easily register your own Namespace at slot 1 instead if you wish to provide different/additional nodes.

In my experience with OPC-UA servers, such as Kepware etc..., all the nodes for configured devices and whatever else they expose are in namespace 2.

closed 2016-07-13 by kevinherron