eclipse-milo/milo #17: Implement some way to "clone" builders

ctron, 2016-07-11:
Setting up a builder like `OpcUaClientConfigBuilder` takes "a few" parameters and method calls.

It would be nice to have either a "clone" method or a "copy constructor" in the builder pattern so that there is no need to do it all over again when only one parameter needs to be changed but you want to keep a mutable instance.

kevinherron, 2016-07-11:
What if there was a `copy()` method hanging off `OpcUaClientConfig` instead which returned a `OpcUaClientConfigBuilder` with all of the values pre-set from the config. This way you're not encouraged to leave some reference to a mutable, non-thread-safe object like a builder laying around.

kevinherron, 2016-07-11:
I'm implementing this idea on `UaTcpStackClientConfig` and `OpcUaClientConfig`.

ctron, 2016-07-11:
Actually I just did it for the server config because I needed them most :-)

ctron, 2016-07-11:
So I will stick with my local changes and wait for your fix! Thanks!!

Just to let you know what I did. I created two additional constructors, beside the default one. Taking in "Config" and "ConfigBuilder" each. The "ConfigBuilder" just calls "this ( other.build() )", which may not be that performant, but this should not be called that often either.

But that allows one to clone the builder and also create a mutable version from an existing configuration again.

True, every type used by the configuration, should be immutable as well.

kevinherron, 2016-07-12:
So I've done half the work... added `copy()` methods to the client config objects. It's someone tedious and boring, especially the tests, so I'm going to take a break and implement the server ones later... :stuck_out_tongue_closed_eyes:

kevinherron, 2016-07-12:
Just missing tests for the server config objects now.

closed 2016-07-19 by kevinherron