eclipse-milo/milo #20: Wiki Update

amitjoy, 2016-07-12:
It would be really helpful if you can update wiki on how to use the library. I would also suggest you to add your example application to this repo as well. It would definitely be helpful to other developers.

kevinherron, 2016-07-12:
That's part of the plan! :smile:

jpposteuca, 2016-10-18:
Is there any other place to get documentation on how to use the library?  I've started working with the sample OPC server, but I'm not sure what all the code does.  For instance, the SubscriptionModel is a bit mysterious.  Thanks in advance!

kevinherron, 2016-10-19:
@jpposteuca documentation is still largely a TODO, unfortunately.

As for `SubscriptionModel`... that's just a naive implementation of the methods needed to implement `MonitoredItemManager` so the example server can handle subscriptions. What makes it especially naive is it just delegates out to the read service calls. In the "real world" you would ideally have some way of actually subscribing to this data.

closed 2019-02-18 by kevinherron