eclipse-milo/milo #23: Resource Leak: SocketServers don't get shut down

ctron, 2016-07-19:
Create a new OPC UA server instance and shutting it down again won't shutdown the underlying SocketServer instance.

ctron, 2016-07-19:
This seem to cause two issues for me:
a) Shutting down a server will trigger the client to throw "unknown endpoint" exceptions
b) Running with different class loaders causes "Address in use" exceptions although servers are shut down

So this may seem a bit esoteric, but in fact it is not. a) is easily triggered when you activate Milo as an optional component and then later deactivate it. The port is kept open and the server log will record "unkown endpoint" exceptions although the service should be disposed. b) Is triggered when you are running unit tests, like with surefire, which puts unit tests in different class loaders. So the "static" allServers collections is actually instantiated multiple times.

kevinherron, 2016-07-21:
@ctron [the PR](https://github.com/eclipse/milo/pull/29) includes a test, but maybe you can verify this fix works in your use case?

ctron, 2016-07-21:
I can give it a try. I have a reproducible case with Maven and unit tests.

ctron, 2016-07-21:
:+1: I can confirm that this works.

Here is what I did:
* switch to branch
* `mvn clean install`
* remove my workaround -> fork all unit tests
* run my unit tests -> OK
* switch back to master branch
* `mvn clean install` -> old version is back
* run my unit tests -> failure: "Address already in use"
* re-apply my workaround
* run my unit tests -> OK

closed 2016-07-21 by kevinherron