eclipse-milo/milo #30: Possible DelegateRegistry race condition?

kevinherron, 2016-07-21:
Reported and discussed here: https://github.com/digitalpetri/ua-client-sdk/issues/9

closed 2016-07-23 by kevinherron

amitjoy, 2016-08-10:
I'm getting the same error:

`org.eclipse.milo.opcua.stack.core.UaSerializationException: no decoder registered for encodingId=NodeId{ns=0, id=449}
	at org.eclipse.milo.opcua.stack.core.serialization.DelegateRegistry$Instance.getDecoder(DelegateRegistry.java:210)
	at org.eclipse.milo.opcua.stack.core.serialization.binary.BinaryDecoder.decodeMessage(BinaryDecoder.java:366)
	at org.eclipse.milo.opcua.stack.client.handlers.UaTcpClientMessageHandler.lambda$onOpenSecureChannel$15(UaTcpClientMessageHandler.java:411)
	at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$decode$1(SerializationQueue.java:58)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$PollAndExecute.run(ExecutionQueue.java:107)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)`

kevinherron, 2016-08-10:
@amitjoy do you have a repeatable example you can share? I've still never seen this reproduced where it wasn't just a ClassLoader issue

amitjoy, 2016-08-10:
In OSGi environment, it happens if the specific bundle classloader is not delegated to the stack. I actually forgot that the we had discussed long back on delegating OSGi classloader to the stack so that OSGi can load the relevant encoders and decoders in runtime. Found it again while going though our previous discussion. :)

Henry198, 2016-09-12:
Hello,

I've this problem with execution (Run As --> Eclipse Application) of my program (that uses some of my Eclipse plugins).

```
[ua-shared-pool-0] ERROR org.eclipse.milo.opcua.stack.client.handlers.UaTcpClientMessageHandler - Error decoding OpenSecureChannelResponse: no decoder registered for encodingId=NodeId{ns=0, id=449}
org.eclipse.milo.opcua.stack.core.UaSerializationException: no decoder registered for encodingId=NodeId{ns=0, id=449}
	at org.eclipse.milo.opcua.stack.core.serialization.DelegateRegistry$Instance.getDecoder(DelegateRegistry.java:210)
	at org.eclipse.milo.opcua.stack.core.serialization.binary.BinaryDecoder.decodeMessage(BinaryDecoder.java:366)
	at org.eclipse.milo.opcua.stack.client.handlers.UaTcpClientMessageHandler.lambda$6(UaTcpClientMessageHandler.java:411)
	at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$1(SerializationQueue.java:58)
	at org.eclipse.milo.opcua.stack.core.util.ExecutionQueue$PollAndExecute.run(ExecutionQueue.java:107)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
java.util.concurrent.ExecutionException: UaException: status=Bad_ConnectionClosed, message=connection closed
	at java.util.concurrent.CompletableFuture.reportGet(Unknown Source)
	at java.util.concurrent.CompletableFuture.get(Unknown Source)
	at org.my.package.communication.opcua.OpcUaClientRunner.createClient(OpcUaClientRunner.java:55)
	at org.my.package.communication.opcua.OpcUaClientRunner.run(OpcUaClientRunner.java:99)
	at java.lang.Thread.run(Unknown Source)
Caused by: UaException: status=Bad_ConnectionClosed, message=connection closed
	at org.eclipse.milo.opcua.stack.client.handlers.UaTcpClientMessageHandler.channelInactive(UaTcpClientMessageHandler.java:160)
	at io.netty.channel.ChannelHandlerInvokerUtil.invokeChannelInactiveNow(ChannelHandlerInvokerUtil.java:56)
	at io.netty.channel.DefaultChannelHandlerInvoker.invokeChannelInactive(DefaultChannelHandlerInvoker.java:93)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:342)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelInactive(AbstractChannelHandlerContext.java:124)
	at io.netty.handler.codec.ByteToMessageDecoder.channelInputClosed(ByteToMessageDecoder.java:349)
	at io.netty.handler.codec.ByteToMessageDecoder.channelInactive(ByteToMessageDecoder.java:309)
	at io.netty.handler.codec.ByteToMessageCodec.channelInactive(ByteToMessageCodec.java:118)
	at io.netty.channel.ChannelHandlerInvokerUtil.invokeChannelInactiveNow(ChannelHandlerInvokerUtil.java:56)
	at io.netty.channel.DefaultChannelHandlerInvoker.invokeChannelInactive(DefaultChannelHandlerInvoker.java:93)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelInactive(AbstractChannelHandlerContext.java:342)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelInactive(AbstractChannelHandlerContext.java:124)
	at io.netty.channel.DefaultChannelPipeline.fireChannelInactive(DefaultChannelPipeline.java:1060)
	at io.netty.channel.AbstractChannel$AbstractUnsafe$7.run(AbstractChannel.java:733)
	at io.netty.util.concurrent.SingleThreadEventExecutor.runAllTasks(SingleThreadEventExecutor.java:339)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:356)
	at io.netty.util.concurrent.SingleThreadEventExecutor$5.run(SingleThreadEventExecutor.java:742)
	... 1 more
```

My context in Eclipse Mars version :
I've a plugin A that is used by a plugin B (A<-B).
I've the plugin B that is used by my main program C (B<-C).
At final : A<-B<-C.

I've execute a first time my program C with "Run as --> Java Application". Everything is working normally :-)
I've execute a second time my program C with "Run as --> Eclipse Application". This is where I get the above error :-s

Any ideas ?

Thanks in advance !

kevinherron, 2016-09-12:
Last night I merged a PR that makes DelegateRegistry not rely on classpath scanning any more - this should fix any weird classloading issues like this.

If you pull and run `mvn install` or otherwise force your project to look for the latest snapshot version you should be able to use the fix.

Henry198, 2016-09-12:
RESOLVED (see https://github.com/eclipse/milo/issues/39)

Thanks very well ! Now, it's working with success !