eclipse-milo/milo #32: Create CI job at HIPP instance

ctron, 2016-07-22:
I do think it is time to create a Hudson (yes I know) job at your HIPP instance.

This would be a simple "Invoke Maven 3" job with `clean install deploy`. This would automatically push to the Nexus OSS repository, since you project is already set up for this.

If you like, I can help you setting this up, but you need to create the Hudson job.

kartben, 2016-07-22:
FWIW you can also ask for a "JIPP" (i.e. Jenkins) instance. See https://bugs.eclipse.org/bugs/show_bug.cgi?id=336262#c42

ctron, 2016-07-22:
Cool, I didn't know they went "out of beta".

kevinherron, 2016-07-22:
I've set up a job that runs `clean install deploy` but I'm still trying to figure out how to trigger it automatically.

ctron, 2016-07-22:
Like that:

[image]

ctron, 2016-07-22:
That is from the secondary package drone build: [1]

If you like I can assign you temporary access, so that you can peek.

[1] https://hudson.eclipse.org/package-drone/job/package-drone-secondary-build/

kevinherron, 2016-07-22:
Okay... so it polls. Is it not possible to setup a webhook that triggers it?

kevinherron, 2016-07-22:
Getting somewhere... it's failing to deploy because sonatype says unauthorized.

kevinherron, 2016-07-22:
Got it working...

closed 2016-07-22 by kevinherron

ctron, 2016-07-22:
And you can have a nice shield too: [image]

kevinherron, 2016-07-22:
I'll save that for when it's actually building successfully... there's some kind of bug that is happening when the tests run on the CI server, possibly related to the change that shut down and removed unused SocketServer instances.