eclipse-milo/milo #37: Bad_NodeIdUnknown

nuccy, 2016-08-22:
Hello.
I'm trying to create a java opcua server based on your library. Since there is not much documentation I use the example from your previous project and try to add a SqrtMethod() to HelloWorld of Milo, so here is a modified addNodes of the HelloWorldNamespace.java

`    private void addNodes(UaFolderNode folderNode) {
        for (Object[] os : STATIC_SCALAR_NODES) {
            String name = (String) os[0];
            NodeId typeId = (NodeId) os[1];
            Variant variant = (Variant) os[2];

      	//  UaMethodNode methodNode = new UaMethodNode(server.getNodeManager(), NodeId.parse("ns=2;i=12546"), new QualifiedName(0, "sqrt"), new LocalizedText("en", "sqrt(x)"), Optional.empty(), Optional.of(UInteger.valueOf(0L)), Optional.of(UInteger.valueOf(0L)), true, true);
      	  
      	  
        UaMethodNode methodNode = UaMethodNode.builder(server.getNodeManager())
        .setNodeId(new NodeId(namespaceIndex, "/Methods/sqrt(x)"))
        .setBrowseName(new QualifiedName(namespaceIndex, "sqrt(x)"))
        .setDisplayName(new LocalizedText(null, "sqrt(x)"))
        .setDescription(LocalizedText.english("Returns the correctly rounded positive square root of a double value."))
        .build();

        try {
      	    AnnotationBasedInvocationHandler invocationHandler=AnnotationBasedInvocationHandler.fromAnnotatedObject(server.getNodeManager(),new SqrtMethod());
      	    methodNode.setProperty(UaMethodNode.InputArguments,invocationHandler.getInputArguments());
      	    methodNode.setProperty(UaMethodNode.OutputArguments,invocationHandler.getOutputArguments());
      	    methodNode.setInvocationHandler(invocationHandler);            
            UaVariableNode node = new UaVariableNode.UaVariableNodeBuilder(server.getNodeManager())
                .setNodeId(new NodeId(namespaceIndex, "Digicam/" + name))
                .setAccessLevel(ubyte(AccessLevel.getMask(AccessLevel.READ_WRITE)))
                .setBrowseName(new QualifiedName(namespaceIndex, name))
                .setDisplayName(LocalizedText.english(name))
                .setDataType(typeId)
                .setTypeDefinition(Identifiers.BaseDataVariableType)
                .build();

            node.setValue(new DataValue(variant));

            server.getNodeManager().addNode(node);
            server.getNodeManager().addNode(methodNode);
            folderNode.addOrganizes(node);
            folderNode.addOrganizes(methodNode);
            
            folderNode.addReference(new Reference(folderNode.getNodeId(),Identifiers.HasComponent,methodNode.getNodeId().expanded(),methodNode.getNodeClass(),true));
        	} catch (Exception e) {
        		logger.error("Error creating sqrt() method.",e);
    	    }
        }
    }`

The class for SqrtMethod is:

`public class SqrtMethod {

    private final Logger logger = LoggerFactory.getLogger(getClass());

    @UaMethod
    public void invoke(
            InvocationContext context,

            @UaInputArgument(
                    name = "x",
                    description = "A value.")
            Double x,

            @UaOutputArgument(
                    name = "x_sqrt",
                    description = "The positive square root of x. If the argument is NaN or less than zero, the result is NaN.")
            Out<Double> xSqrt) {

    	System.out.println("sqrt(" + x.toString() + ")");
        logger.debug(String.format("Invoking sqrt() method of Object '%s'", context.getObjectNode().getBrowseName().getName()));

        xSqrt.set(Math.sqrt(x));
    }

}`

To connect to the server I use Prosys OPC UA Java Client and when I try to call the appropriate method with this client I get:
`Failed to call:Bad_NodeIdUnknown (0x80340000) "The node id refers to a node that does not exist in the server address space."`

Since variables work correctly and I never had any issues with Prosys I assume something is wrong with your software. Can you please help to solve it? Thanks.

kevinherron, 2016-08-22:
`ExampleNamespace` on the `more-examples` branch has an example implementation of the sqrt method now: https://github.com/eclipse/milo/blob/more-examples/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/server/ExampleNamespace.java

The part you may have missed is overriding `getInvocationHandler`, which you can see at the bottom of the example namespace.

nuccy, 2016-08-22:
Thanks. Was not obvious for me and I missed more examples you've provided.

kevinherron, 2016-08-22:
The more-examples branch is in progress. Once I finish up a few more things I'll merge it back into master and it won't be so hidden away.

closed 2016-09-01 by nuccy