eclipse-milo/milo #39: OpenSecureChannelResponse: no decoder registered for encodingId=NodeId{ns=0, id=449}

williamschey, 2016-09-07:
When running the client library from a JAR in Ubuntu 16.04.1 within Oracle 1.8.0_101-b13 I get:

> ERROR o.e.m.o.s.c.h.UaTcpClientMessageHandler - Error decoding OpenSecureChannelResponse: no decoder registered for encodingId=NodeId{ns=0, id=449}
> org.eclipse.milo.opcua.stack.core.UaSerializationException: no decoder registered for encodingId=NodeId{ns=0, id=449}

This only occurs when running the library from the JAR, it does NOT occur if I run the library from the debugger (i.e. individually referencing the jars from the maven directory, it works). 

Delving into the issue, it seems to relate to:
    classPath.getTopLevelClasses("org.eclipse.milo.opcua.stack.core.types.structured");
in DelegateRegistry.

After a little more digging, I saw that the Guava ClassPath is annotated with @Beta. I then went back to check the doco and found:

> "It's worth noting that ClassPath is a best-effort utility. It only scans classes in jar files or under a file system directory. Neither can it scan classes managed by custom class loaders that aren't URLClassLoader. So don't use it for mission critical production tasks."

I am not using a custom class loader so decided to look into this class. Long story short, it seems quite brittle. So I did a little research and found https://github.com/ddopson/java-class-enumerator which is under a WTFPL license :-).

I have done a fix which I will commit and do a pull request. Let me know if you need any additional info.

kevinherron, 2016-09-07:
Thanks for taking the time to write this up and submit a fix. I won't be able to merge it directly unless you have an Eclipse CLA on file, but I think the most important thing you've done is potentially identify what's been causing this issue for people.

I'm going to take a step back and evaluate whether to continue using a classpath scanning approach, likely using something like you've implemented, or to just simplify things and load all these structures in a more traditional manner.

williamschey, 2016-09-07:
Hi Kevin,

Eclipse CLA completed.

Let me know if you need anything else.

Regards,
Will

kevinherron, 2016-09-09:
@williamschey can you checkout the `no-classpath-scanning` branch and verify it works for you?

williamschey, 2016-09-11:
Thanks @kevinherron, all fixed. Do you know when you'll merge and release?

kevinherron, 2016-09-11:
@williamschey I'll merge and release a new snapshot tonight. Releasing a final version of 0.1.0 is a bit of a process since it will be the project's first release. It's underway though...

williamschey, 2016-09-11:
Shapshot will be absolutely fine. Thanks again.

kevinherron, 2016-09-11:
Merged and deployed. Thanks for helping!

closed 2016-09-11 by kevinherron