eclipse-milo/milo #43: How to convert ExpandedNodeId to NodeId ?

AriSuutariST, 2016-09-14:
I'm trying to browse recursively the server address space using OpcUaClient.browse(). I cannot figure out what is the correct way to create a NodeId from ExpandedNodeId that is returned in BrowseResult.getReferences().

Things seem to work (with Kepware) when I use NodeId.parse() with suitable string created
from ExpandedNodeId values, but I think that is not the correct way to do it.

An example for browsing the server address space in client would be great.

kevinherron, 2016-09-14:
`ExpandedNodeId.local()` will give you an `Optional<NodeId>`. You can check if it's "local" with the `isLocal()` method.

An `ExpandedNodeId` that isn't "local" is rare - It means the node actually resides in *another* server, identified by looking at the URI of the server at `serverIndex` in the server's `ServerArray` node... But, to cover all cases, many UA structures return `ExpandedNodeId` instead of`NodeId` so it's just a nuisance you can deal with. 

I'll see about getting some better browse examples going.

AriSuutariST, 2016-09-15:
Thanks, this works perfectly.

closed 2016-09-15 by AriSuutariST