eclipse-milo/milo #49: XMLDecoder.decodeExpandedNodeId not implemented

comtel2000, 2016-09-16:
Hi,
I tried to use the internal XmlEncoder/XmlDecoder function to serialize and unserialize ReferenceDescription objects. Not sure if there is a better way but my work around fails on
org.eclipse.milo.opcua.stack.core.serialization.xml.XmlDecoder.XmlDecoder.decodeExpandedNodeId which returns null.

```java
    String sample =
        "<ReferenceDescription><ReferenceTypeId><Identifier>ns=0;i=46</Identifier></ReferenceTypeId><IsForward>true</IsForward><NodeId><Identifier>svr=0;i=2994</Identifier></NodeId><BrowseName><NamespaceIndex>0</NamespaceIndex><Name>Auditing</Name></BrowseName><DisplayName><Locale></Locale><Text>Auditing</Text></DisplayName><TypeDefinition><Identifier>svr=0;i=68</Identifier></TypeDefinition></ReferenceDescription>";
    XmlDecoder decoder = new XmlDecoder(new StringReader(sample));
    // skip root element
    decoder.decodeQualifiedName("<>");
    ReferenceDescription rd = ReferenceDescription.decode(decoder);

    assertNotNull(rd);
    assertNotNull(rd.getTypeId());
    assertEquals(Boolean.TRUE, rd.getIsForward());
    // not decoded
    assertNotNull(rd.getNodeId());
```

kevinherron, 2016-09-16:
`XmlEncoder` and `XmlDecoder` are not fully implemented or supported yet.

comtel2000, 2016-09-16:
ok, do you accept pull request for that? I still search for a easy way to store subscription lists and the xml format seems to be a perfect solution.

kevinherron, 2016-09-16:
Sure, just make sure you have an Eclipse CLA on file before you submit the PR.

It's a fair bit of work and testing required to fully implement `XmlEncoder` and `XmlDecoder` and I've found the documentation and examples in the UA spec to be lacking when it comes to the XML encoding. It's not even implemented in the Java stack maintained by the OPC foundation.

closed 2016-09-27 by kevinherron