eclipse-milo/milo #52: strictEndpointUrlsEnabled doesn't work anymore

cjmaloof, 2016-09-19:
It used to be that SocketServers.getServer(), in the case that there was just one server on a host, would return that server even if no path is specified, as long as strictEndpointUrlsEnabled was false.  That behavior seems to have gone away, but the documentation on UaTcpStackServerConfig still seems to say it should work.

kevinherron, 2016-09-20:
Hmm. I'll take a look at adding this behavior back. Thanks.

kevinherron, 2018-05-23:
Long overdue update... this setting is going to get removed and the functionality will not come back. Multiple interoperability events have been attended since this stopped working and there have been no problems with other clients connecting. The only ever known issue was with some old version of the Prosys client which was fixed long ago.

closed 2018-05-23 by kevinherron