eclipse-milo/milo #57: Write operation fails with the message Bad_WriteNotSupported

tosuns, 2016-10-02:
After a while I started again with the evaluation of this library. My previous issue seems to be resolved.
But now when I try to change a value on the Beckhoff-OPCUA-Server it fails with the message Bad_WriteNotSupported, even if the variable is set to readable and writable. I tried the Beckhoff OPCUA-Client and the write operation worked fine and I'm able to read the value of the variable with the milo opcua client (read-operation works fine, but not the write operation).

The Test-case in the milo project works fine too. So I'm not sure what could actually be the cause for this issue.

My Test-case looks as follows: 

`Test
    public void lowlevelTest() throws InterruptedException,
                                      ExecutionException {

        final OpcUaClient opcUaclient = UaTcpStackClient.getEndpoints("opc.tcp://localhost:4840")
                .thenApply(endPoints -> {
                    EndpointDescription endpoint = Arrays.stream(endPoints)
                            .filter(e -> e.getSecurityPolicyUri().equals(SecurityPolicy.None.getSecurityPolicyUri()))
                            .findFirst()
                            .orElseThrow(() -> new RuntimeException("no desired endpoints return from the opcua server!"));
                    OpcUaClientConfig clientConfig = OpcUaClientConfig.builder()
                            .setApplicationName(LocalizedText.english("test opcua client"))
                            .setApplicationUri("urn:test:opcua:client")
                            .setEndpoint(endpoint)
                            .setRequestTimeout(uint(60000))
                            .setExecutor(Executors.newCachedThreadPool())
                            .build();
                    return new OpcUaClient(clientConfig);
                })
                .get();
        final NodeId nodeId = new NodeId(7, "GlobalData.AirConditioning.Config.T_on");
        final DataValue before = opcUaclient.readValue(0, TimestampsToReturn.Both, nodeId)
                .get();
        System.out.println(MessageFormat.format("before write value: {0}", before));
        
        final StatusCode statusCode = opcUaclient.writeValue(nodeId, new DataValue(new Variant(32f))).get();

        Assert.assertTrue(statusCode.toString(), statusCode.isGood());

        final DataValue after = opcUaclient.readValue(0, TimestampsToReturn.Both, nodeId)
                .get();

        System.out.println(MessageFormat.format("after write value: {0}", after));
        
        Assert.assertThat(after.getValue().getValue(), not(is(before.getValue().getValue())));
    }`

The assertion regarding the StatusCode fails. 

`java.lang.AssertionError: StatusCode{name=Bad_WriteNotSupported, value=0x80730000, quality=bad}
	at org.junit.Assert.fail(Assert.java:93)
	at org.junit.Assert.assertTrue(Assert.java:43)
	at com.milo.opcua.MiloOpcUaClientNGTest.lowlevelTest(MiloOpcUaClientNGTest.java:151)`

I'm testing the most recent master branch.

kevinherron, 2016-10-02:
The constructor you're using for DataValue takes a value and uses
StatusCode.GOOD by default. If I remember correctly, the Beckhoff servers
are very picky about allowing writes that include a StatusCode. Try
building your DataValue using the full 4-argument constructor but supplying
null values for everything but the value argument.

tosuns, 2016-10-02:
Thanks, that worked 👍

closed 2016-10-02 by kevinherron