eclipse-milo/milo #61: "sdk-server" exports "org.eclipse.milo.opcua.sdk.core"

ctron, 2016-10-14:
As @JulianSchuette explained issue #59 the "sdk-server" project exports "org.eclipse.milo.opcua.sdk.core". While this doesn't matter in plain Java, it matters when using OSGi. Since each package should only be exported by one single bundle (aka JAR).

There two workarounds to this (like using bundle imports or going with fragments) however this seems overkill to me.

@kevinherron Could we either rename the package "org.eclipse.milo.opcua.sdk.core" or ".server" or move it into the "sdk-core" package.

ctron, 2016-10-14:
@JulianSchuette can I reach you somewhere on IRC or something similar. You can drop me an e-mail at "jreimann@redhat.com" or "ctron@dentrassi.de" if you like.

kevinherron, 2016-10-14:
At first glance it seems the remaining 6 classes in the `...sdk.core` package in `sdk-server` module can be moved to the `sdk-core` module. 

Whether or not that's the appropriate location for all of them is another thing... It looks like `NumericRange` is the only one that may be of no use to a client.

ctron, 2016-10-14:
So .. would it hurt to do that? ... Leaving it as it is would ... from an OSGi perspective of course :wink:

kevinherron, 2016-10-14:
I think it's pretty safe and easy.

kevinherron, 2016-10-14:
So easy... that I will do it real quick.

kevinherron, 2016-10-14:
Can one of you checkout that branch and verify or let me know if anything else needs moving.

ctron, 2016-10-14:
Give me an hour. I will try it.

closed 2016-10-14 by kevinherron

JulianSchuette, 2016-10-14:
Ah, great! Thank you guys for the super quick fix! I'll let you know if I run in any further OSGi-/Karaf-shallows :-)