eclipse-milo/milo #66: BindAddress listening on wrong IP

Pro, 2016-10-17:
In the `OpcUaServerConfig` you can set the bind address with:

(simplified as array instead of list)
`.setBindAddresses([])`

If I use `setBindAddresses(["localhost"])` the server listens on:

```
[main] INFO org.domain.UaServer - startDiscoveryServer()
[main] INFO org.eclipse.milo.opcua.sdk.server.namespaces.OpcUaNamespace - Loaded nodes in 418ms.
[main] INFO org.eclipse.milo.opcua.sdk.server.NamespaceManager - added namespace index=0, uri=http://opcfoundation.org/UA/
[main] INFO org.eclipse.milo.opcua.sdk.server.NamespaceManager - registered and added namespace index=1, uri=urn:domain:opcua:discovery:server
[main] INFO org.eclipse.milo.opcua.sdk.server.OpcUaServer - Binding endpoint opc.tcp://my-host:4840 to localhost [None/None]
[main] INFO org.eclipse.milo.opcua.sdk.server.OpcUaServer - Binding endpoint opc.tcp://localhost:4840 to localhost [None/None]
[main] INFO org.eclipse.milo.opcua.sdk.server.OpcUaServer - Binding endpoint opc.tcp://127.0.0.1:4840 to localhost [None/None]
[main] INFO org.eclipse.milo.opcua.sdk.server.OpcUaServer - eclipse milo opc-ua stack version: dev
[main] INFO org.eclipse.milo.opcua.sdk.server.OpcUaServer - eclipse milo opc-ua sdk version: dev
[main] INFO org.eclipse.milo.opcua.sdk.server.NamespaceManager - registered and added namespace index=2, uri=urn:eclipse:milo:opcua:test-namespace
[main] INFO org.eclipse.milo.opcua.stack.server.tcp.UaTcpStackServer - opc.tcp://my-host:4840 bound to /127.0.0.1:4840 [None/None]
[main] INFO org.eclipse.milo.opcua.stack.server.tcp.UaTcpStackServer - opc.tcp://localhost:4840 bound to /127.0.0.1:4840 [None/None]
[main] INFO org.eclipse.milo.opcua.stack.server.tcp.UaTcpStackServer - opc.tcp://127.0.0.1:4840 bound to /127.0.0.1:4840 [None/None]

```

As you can see it also listens on `opc.tcp://my-host:4840` but bound to `127.0.0.1:4840`.

Imho this is a wrong behavior, because then the endpoint discovery URLs also contain this entry, but if then use a connect to this URL, it can't connect, because it doesn't take the loopback on linux, but eth0, and since it is bound to 127.0.0.1, it can't connect.

The correct way here would be to either do not add this entry (because the developer said that he only wants to listen to localhost) or bind it to 0.0.0.0:4840

The same happens if I use `setBindAddresses(["my-host"])`, here it should definitely also bind to 0.0.0.0:4840

kevinherron, 2016-10-17:
I think the best thing I could do here is make it so `OpcUaServer#getHostnames(String)` is user-configurable. 

I don't think there's any implementation we could come up with that would satisfy every use case.

For example, I can see your logic about the canonical hostname being included in the endpoints being confusing since we're bound to localhost, but without including it that would prevent an OPC-UA client application running on the same machine from accessing the server at `my-host:4840`, which seems overly strict to me.

kevinherron, 2016-10-17:
My original thought when writing `getHostnames()` was to lean towards being lenient/flexible, including as many hostnames in the endpoints as I reasonably could.

In the real world we've found that the majority of servers end up being configured (or misconfigured...) with either incorrect or unreachable endpoints, which has lead to pretty much all UA client application implementations to have logic that replaces the hostname in the endpoints provided by servers with the hostname/address used to get the endpoints in the first place.

Pro, 2016-10-17:
I just ran into the problem using open62541 and I think that I identified the problem.

You are using the value of `bindAddress` to create the socket here:
https://github.com/eclipse/milo/blob/master/opc-ua-stack/stack-server/src/main/java/org/eclipse/milo/opcua/stack/server/tcp/UaTcpStackServer.java#L148

`String bindAddress = endpoint.getBindAddress().orElse(endpointUri.getHost());`

I just hacked it to test with:

`String bindAddress = endpointUri.getHost();`

and tadaaa, it works. In ubuntu the local IP for the hostname is `127.0.1.1` compared to localhost `127.0.0.1`

```
[main] INFO org.domain.UaServer - startDiscoveryServer()
[main] INFO org.eclipse.milo.opcua.sdk.server.namespaces.OpcUaNamespace - Loaded nodes in 589ms.
[main] INFO org.eclipse.milo.opcua.sdk.server.NamespaceManager - added namespace index=0, uri=http://opcfoundation.org/UA/
[main] INFO org.eclipse.milo.opcua.sdk.server.NamespaceManager - registered and added namespace index=1, uri=urn:domain:opcua:discovery:server
[main] INFO org.eclipse.milo.opcua.sdk.server.OpcUaServer - Binding endpoint opc.tcp://my-host:4840 to localhost [None/None]
[main] INFO org.eclipse.milo.opcua.sdk.server.OpcUaServer - Binding endpoint opc.tcp://localhost:4840 to localhost [None/None]
[main] INFO org.eclipse.milo.opcua.sdk.server.OpcUaServer - Binding endpoint opc.tcp://127.0.0.1:4840 to localhost [None/None]
[main] INFO org.eclipse.milo.opcua.sdk.server.OpcUaServer - eclipse milo opc-ua stack version: dev
[main] INFO org.eclipse.milo.opcua.sdk.server.OpcUaServer - eclipse milo opc-ua sdk version: dev
[main] INFO org.eclipse.milo.opcua.sdk.server.NamespaceManager - registered and added namespace index=2, uri=urn:eclipse:milo:opcua:test-namespace
[main] INFO org.eclipse.milo.opcua.stack.server.tcp.UaTcpStackServer - opc.tcp://my-host:4840 bound to /127.0.1.1:4840 [None/None]
[main] INFO org.eclipse.milo.opcua.stack.server.tcp.UaTcpStackServer - opc.tcp://localhost:4840 bound to /127.0.0.1:4840 [None/None]
[main] INFO org.eclipse.milo.opcua.stack.server.tcp.UaTcpStackServer - opc.tcp://127.0.0.1:4840 bound to /127.0.0.1:4840 [None/None]
```

Is there a specific reason why you are using `getBindAddress()` instead of `endpointUrl.getHost()`?

kevinherron, 2016-10-17:
I think I set a bad example in the `ExampleServer` implementation by using "localhost" as a bind address. 

The intention is that bind addresses are always configured as IP addresses and not as a hostnames because I want the addresses the server is configured to bind to to be un-ambiguous. 

From there, the server tries to intelligently figure out what hostnames and endpoint URLs to use (the getHostnames() function I'm talking about making extensible).

Pro, 2016-10-17:
Hmm, maybe you could ad a check if the value of `setBindAddresses([])` is a hostname or IP and then warn the user.
But I think that the cleaner solution is to use the endpoint.getHost() to bind to, because then the OS has to figure out the correct ip.

And yes, the example should be changed then :)

kevinherron, 2016-10-17:
> But I think that the cleaner solution is to use the endpoint.getHost() to bind to, because then the OS has to figure out the correct ip.

The whole point is I don't want the OS to figure it out. In a system with multiple NICs I need to be able to be very specific about which address I'm binding to, and then there's room to be less strict on top of that by allowing more than one endpoint URL to point to that NIC.

So I think this is making my point about there not being one solution that will please all... I'll figure out how to make this more configurable.

kevinherron, 2016-10-29:
@Pro if you get a chance, check out this PR/branch and see if you can accomplish what you need with it. I'll leave the PR open for a few days before merging it.

closed 2016-11-07 by kevinherron