eclipse-milo/milo #77: Reconnect attempts stops already on the first try

tosuns, 2016-11-02:
I'm currently trying to handle the resubscribtion with either `SessionActivityListener` or `SubscribtionListener` but after my tests I noticed that the reconnction logic is not as reliable as I would expect it to be. I would expect the client would always retry to reconnect to the server, does not matter what kind of exception occurs. 

Sometimes the reconnection works but most of the time it doesn't.

What I did: 

1. connected the client to the server
2. created a subscribtion with monitored items
3. restarted the opcua server
4. getting the following message from the console:
```
Fein:   Scheduling reconnect for +0 seconds...
Information:   onSessionInActive
Fein:   0 seconds elapsed; reconnecting...
Fein:   Channel bootstrap failed: Connection refused: no further information: <Computer>/<ipaddress>4840
java.net.ConnectException: Connection refused: no further information: <Computer>/<ipaddress>4840
	at sun.nio.ch.SocketChannelImpl.checkConnect(Native Method)
	at sun.nio.ch.SocketChannelImpl.finishConnect(SocketChannelImpl.java:717)
	at io.netty.channel.socket.nio.NioSocketChannel.doFinishConnect(NioSocketChannel.java:224)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.finishConnect(AbstractNioChannel.java:289)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:545)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:485)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:399)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:371)
	at io.netty.util.concurrent.SingleThreadEventExecutor$2.run(SingleThreadEventExecutor.java:112)
	at java.lang.Thread.run(Thread.java:745)

Fein:   Reconnect failed: Connection refused: no further information: <Computer>/<ipaddress>4840
java.net.ConnectException: Connection refused: no further information: <Computer>/<ipaddress>4840
	at sun.nio.ch.SocketChannelImpl.checkConnect(Native Method)
	at sun.nio.ch.SocketChannelImpl.finishConnect(SocketChannelImpl.java:717)
	at io.netty.channel.socket.nio.NioSocketChannel.doFinishConnect(NioSocketChannel.java:224)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.finishConnect(AbstractNioChannel.java:289)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:545)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:485)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:399)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:371)
	at io.netty.util.concurrent.SingleThreadEventExecutor$2.run(SingleThreadEventExecutor.java:112)
	at java.lang.Thread.run(Thread.java:745)

Fein:   disconnect complete, state set to Idle
Fein:   Channel bootstrap failed: Connection refused: no further information: <Computer>/<ipaddress>4840
java.net.ConnectException: Connection refused: no further information: <Computer>/<ipaddress>4840
	at sun.nio.ch.SocketChannelImpl.checkConnect(Native Method)
	at sun.nio.ch.SocketChannelImpl.finishConnect(SocketChannelImpl.java:717)
	at io.netty.channel.socket.nio.NioSocketChannel.doFinishConnect(NioSocketChannel.java:224)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.finishConnect(AbstractNioChannel.java:289)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:545)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:485)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:399)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:371)
	at io.netty.util.concurrent.SingleThreadEventExecutor$2.run(SingleThreadEventExecutor.java:112)
	at java.lang.Thread.run(Thread.java:745)
```
Afterwards neither the SubscribtionListener or the SessionActivityListener get called. 

I guess the issue could be caused by a dead lock. But I'm not sure whether the cause is the milo-sdk or my environment. The reason why I think it is a kind of dead lock is, I deployed my client to a javaee application server, when I noticed the dead lock behaviour I disabled the client and then the log printed out the session active and subscribtion transfer failed events.

kevinherron, 2016-11-02:
Automatic reconnecting should work basically "hands off". The only callback you really need to implement is the one that notifies you a transfer failed.

Can you turn the loggers for `ClientChannelManager` and `ClientSessionManager` to DEBUG and then run your test again?

kevinherron, 2016-11-13:
The reconnect logic has been stressed pretty heavily both in test and production environments. I'm going to assume it's something specific to your environment, maybe something with the container and its executors.

If you want to re-open this, provide logs, thread dumps, etc... feel free.

closed 2016-11-13 by kevinherron