eclipse-milo/milo #78: Performance Test

Ox24, 2016-11-04:
Hi,

are there any performance tests for OPC UA in general (and for your implemantion of the opc ua stack)?

Greetings

kevinherron, 2016-11-04:
Some time ago I had put together a little benchmarking client that could be run against any server. Basically, all it did was stress the stack's ability to encode/decode by issuing Read requests at varying concurrency and connection levels.

Is there something in particular you're looking for?

Ox24, 2016-11-04:
Can you provide me with your client? I also want to test what load impact a opc ua connection would have to the network if i try to read like 10k variables :) Maybe i can add ur code to a "framework" to test different readcycles + datatypes + task etc.

kevinherron, 2016-11-04:
I'll see if I can find it. Doesn't look like it's in any of my private repos, might have to look on some old external drives.

kevinherron, 2016-11-04:
Here you go: https://github.com/digitalpetri/ua-bench

Like I said, it's not much more than a test of a stack's serialization and concurrency abilities.

closed 2016-11-04 by kevinherron

Ox24, 2016-11-06:
I missed to fork it, can u make it public again? :+1:

kevinherron, 2016-11-06:
Here, I replaced it with an updated version: https://github.com/digitalpetri/stack-bench