eclipse-milo/milo #88: Issue while running as a docker container

harrybvp, 2016-11-18:
I am trying to dockerize OPC UA Server implementation
I have a OPC UA client running on a different host .

However, i am facing issues with the setBindAddresses.
When i am using 0.0.0.0 , following binding is taking place
```
Binding endpoint opc.tcp://66466b702420:12687 to 0.0.0.0 [None/None]
Binding endpoint opc.tcp://localhost:12687 to 0.0.0.0 [None/None]
Binding endpoint opc.tcp://172.18.0.4:12687 to 0.0.0.0 [None/None]
Binding endpoint opc.tcp://127.0.0.1:12687 to 0.0.0.0 [None/None]
```

172.18.0.4 is docker container ip (obviously not accessible outside host)

As you can see all binding ips are internal ips and not accessible outside host.
So OPC UA Client is not able to discover OPC UA Server.

Only way i get it working is pass container name (container host name) in the setBindAddresses
then the binding becomes
`Binding endpoint opc.tcp://containername:12687 to containername [None/None]
`and on the host which is running OPC UA Client  . i added a entry on /etc/hosts/ 
mapping containername to <ip of OPC UA Server Host IP>

I  am hoping for a more cleaner solution?

kevinherron, 2016-11-18:
I'm not a docker expert but it seems this is basically the intended networking setup for a container. Communication is done with the host via port forwarding and the container only ever sees its internal IP address.

If you have some way of knowing your externally reachable IP address from inside the server you can call `setHostname(String)` while building the `OpcUaServerConfig` and whatever value you provide there will be used in one of the sets of endpoints when the server starts up.

kevinherron, 2016-11-18:
The other possibility is that the OPC-UA client on the other side should just be smart enough to handle this without resorting to an entry in /etc/hosts.

I just answered a question on SO about how you might do this in the Milo client: http://stackoverflow.com/questions/40554198/java-opc-ua-client-eclipse-milo-endpoint-url-changes-to-localhost

Most other UA clients have similar functionality.

harrybvp, 2016-11-18:
I was aware of port forwarding.
OPC UA client was able to see OPC UA server using public ip of host which is running OPC UA server docker container and forwared port but was complaining no UA endpoints due to mismacth b/w IP of host and  'endpointUrl' returned by OPC UA Server.

However passing public ip(in my case OPC UA Server's Host)  via setHostName did the trick.
It is indeed setting one of the endpoints with public ip in it and binding to 0.0.0.0 which is great.

Thanks a lot for your suggestion!

kevinherron, 2016-11-18:
Yep, no problem. 

Ideally there would be some configuration in the client for handling this situation also.

There are a few corner cases such as docker containers and servers in the cloud behind firewall/port-forwarding/etc where it may not be possible to get the server to return a publicly reachable endpoint URL to all clients.

closed 2016-11-18 by kevinherron