eclipse-milo/milo #92: Problems when session lifetime has expired

AriSuutariST, 2016-11-23:
When session lifetime expires (ie. there is "Session id=NodeId{ns=1, id=Session:0841d805-7944-40f3-a7b3-b710d938d77a} lifetime expired (120000ms)" on debug log) the next operation seems to run into problems.

I observed this with readValues(), but it can be reproduced with ReadExample by adding Thread.sleep(130000) before readServerStateAndTime call (timeout in ClientExampleRunner needs to be bumped up also).

For asynchronous request, the system seems to notice that session has expired and builds a new one, but the original request never completes (ie. thenAccept is never called). A synchronous request using get() results in "UaServiceFaultException: status=Bad_SessionIdInvalid, message=The session id is not valid."

kevinherron, 2016-11-23:
This is actually just how `CompletableFuture` works. `thenAccept` is only called in the success case. Because it's an example I left it short rather than writing more error handling code. If you modify the example to be something like this you should get the failure callback too:

```java
        Thread.sleep(125000);

        // asynchronous read request
        readServerStateAndTime(client).whenComplete((values, ex) -> {
            if (ex != null) {
                ex.printStackTrace();
                future.completeExceptionally(ex);
            } else {
                DataValue v0 = values.get(0);
                DataValue v1 = values.get(1);

                logger.info("State={}", ServerState.from((Integer) v0.getValue().getValue()));
                logger.info("CurrentTime={}", v1.getValue().getValue());
                future.complete(client);
            }
        });
```

AriSuutariST, 2016-11-23:
Yes, this is ok, I was just confused because after the session has timed out the next request triggers session setup, which made me think that there would be automatic retry for original request. I'll handle this in my own client code.

closed 2016-11-23 by AriSuutariST

kevinherron, 2016-11-23:
@AriSuutariST your client should periodically read the server state and time (or whatever node, really) to keep the session alive.

AriSuutariST, 2016-11-23:
Did just that, works fine now.

shahalamansari, 2020-06-29:
@ASunc  can you please share how you implemented the keep-alive to keep the session alive?

kevinherron, 2020-06-29:
@shahalamansari this hasn't been necessary for a long time. The client SDK automatically reads the server's State variable as a keep alive.