eclipse-milo/milo #100: Read values of datatype 5

Christian-Block, 2016-12-05:
Hi,
i have a method to read different NodeIDs:
```
public DataValue readNode(NodeId nodeID) throws InterruptedException, ExecutionException {
    VariableNode node = client.getAddressSpace().createVariableNode(nodeID);
    DataValue value = node.readValue().get();
    return value;
}
```
My aim is to read the values as String or Integer. The most nodes are ok. But I have some nodes which have an other datatype e.g.:
System.out.println(ua.readNode("node").getValue());
    -->Variant{value=[4]}
System.out.println(ua.readNode("EGX_AktuellWirkleistung_ID4").getValue().getDataType().hashCode());
    --> 5
System.out.println(ua.readNode("EGX_AktuellWirkleistung_ID4").getValue().getValue().toString());
    --> [Lorg.eclipse.milo.opcua.stack.core.types.builtin.unsigned.UShort;@6a2b953e

How can i read values of datatype 5 or rather values in [*] ?

kevinherron, 2016-12-05:
It looks like some of the nodes you're reading have either an array
datatype or in the case of "EGX_AktuellWirkleistung_ID4" the datatype is
UInt16.

Your client needs to be able to handle this. You can't request the server
return a different datatype for you.

Reading the DataType, ValueRank, and ArrayDimensions attributes should be
enough for your client to fully understand what the Value is might be.

kevinherron, 2016-12-06:
Closing this for now. In the future this kind of question (versus reporting a bug) would be better suited for the mailing list: https://dev.eclipse.org/mailman/listinfo/milo-dev

Thanks!

closed 2016-12-06 by kevinherron