eclipse-milo/milo #103: Creating nodes on the server with clients

martingr, 2017-01-07:
I'm trying to set up a very basic and simple client and server application trying to understand how everything works. According to the OPC UA specification the NodeManagement Service Set is used with a client to create new nodes on a server. But I can't find a way to use it. Is it not yet implemented or would i have to implement it on my own? Or is there even another way letting a client create nodes on a server?

kevinherron, 2017-01-07:
`Namespace` implements `NodeManager`, which has all the node management service calls. Its default implementations respond with `Bad_NotSupported`. You would need to override these in your `Namespace` implementation and provide implementations that actually created the nodes.

kevinherron, 2017-01-07:
Questions like about how to use the library are more appropriate for the Gitter channel, mailing list, or Stack Overflow. Gonna close the issue.

closed 2017-01-07 by kevinherron