eclipse-milo/milo #104: OpcUaSubscriptionManager - no item for clientHandle=xxx

JayceJiaa, 2017-01-12:
Hello here, I'm developing our client with your client SDK, 
this problem comes when I tried to create subscription on connecting to the server occasionally. I assuming that it may caused by we have a big amount points in one subscription(about 20,000 data nodes). The error message are bellow. Is there anybody has any idea about what happened to my app or how to make the subscription stable?
`16:33:48.687 [ua-shared-pool-1] WARN  o.e.m.o.s.c.s.OpcUaSubscriptionManager - no item for clientHandle=3052
16:33:48.687 [ua-shared-pool-1] WARN  o.e.m.o.s.c.s.OpcUaSubscriptionManager - no item for clientHandle=3053
16:33:48.687 [ua-shared-pool-1] WARN  o.e.m.o.s.c.s.OpcUaSubscriptionManager - no item for clientHandle=3232
16:33:48.687 [ua-shared-pool-1] WARN  o.e.m.o.s.c.s.OpcUaSubscriptionManager - no item for clientHandle=3055
16:33:48.687 [ua-shared-pool-1] WARN  o.e.m.o.s.c.s.OpcUaSubscriptionManager - no item for clientHandle=3056`

kevinherron, 2017-01-12:
This means the server is sending you PublishResponses containing notifications for items that you either never subscribed to or were previously subscribed but have deleted.

If it's happening continuously there may be an issue with the server. If it happens once, right after deleting the items, it might just be that the responses were already in flight when you deleted the items.

JayceJiaa, 2017-01-13:
But I didn't take any delete action after the subscription established, all I did is that creating a subscription with a list of identifiers after the project initialized just like what you did in the example. It's happening by chance, and it will work if I restart my project. :( I'm really puzzled with this unstable factor~BTW, is there an auto-connect like method could help me maintain the subscriptions? Thanks for your helpful answer!

kevinherron, 2017-01-13:
What server are you connected to? There might not be any problem with your client, it's hard to tell without more information.

I'm going to close this issue. Come by the gitter channel or use the mailing list to continue troubleshooting.

closed 2017-01-13 by kevinherron