eclipse-milo/milo #105: Aggregate OPC UA Servers

Ox24, 2017-02-01:
Is there an example of aggregating opc ua servers with this libary? 
Illustrated like this:

Client     Client    Client
   |             |            |
Aggregation Server
         |                  |
UA Server1 UA Server 2

So u can access data presented by any opc ua server from one server...

kevinherron, 2017-02-01:
There's not an example. Aggregation is actually a lot of work to do right (I think...), and by the time the example was any good you'd practically have a product on your hands.

Ox24, 2017-02-01:
Is there some example from the opc ua foundation? like something defined in their specification?

kevinherron, 2017-02-01:
I'm not aware of anything, but that doesn't mean it doesn't exist :)

Ox24, 2017-02-01:
Ok thx, maybe i try to implement one :)

closed 2017-02-16 by kevinherron