eclipse-milo/milo #106: OPC-UA server problem with certificate?

jbarbosa29, 2017-02-15:
Hi all,

I've a problem connecting to a OPC-UA server. The error is depicted below. Can anyone, help me solving this issue?

`org.eclipse.milo.opcua.stack.core.UaSerializationException: javax.xml.stream.XMLStreamException: Trying to output second root, <ServerCertificate> at org.eclipse.milo.opcua.stack.core.serialization.xml.XmlEncoder.write(XmlEncoder.java:384)[73:org.eclipse.milo.stack-core:0.1.0] at org.eclipse.milo.opcua.stack.core.serialization.xml.XmlEncoder.writeValue(XmlEncoder.java:389)[73:org.eclipse.milo.stack-core:0.1.0] at org.eclipse.milo.opcua.stack.core.serialization.xml.XmlEncoder.encodeByteString(XmlEncoder.java:207)[73:org.eclipse.milo.stack-core:0.1.0] at org.eclipse.milo.opcua.stack.core.types.structured.EndpointDescription.encode(EndpointDescription.java:111)[73:org.eclipse.milo.stack-core:0.1.0] at org.apache.camel.component.milo.client.internal.SubscriptionManager.toString(SubscriptionManager.java:514)[54:de.dentrassi.camel.milo.camel-milo:0.1.0] at org.apache.camel.component.milo.client.internal.SubscriptionManager.performConnect(SubscriptionManager.java:458)[54:de.dentrassi.camel.milo.camel-milo:0.1.0] at org.apache.camel.component.milo.client.internal.SubscriptionManager.connect(SubscriptionManager.java:380)[54:de.dentrassi.camel.milo.camel-milo:0.1.0] at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)[:1.8.0_111] at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_111] at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)[:1.8.0_111] at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)[:1.8.0_111] at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)[:1.8.0_111] at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)[:1.8.0_111] at java.lang.Thread.run(Thread.java:745)[:1.8.0_111] Caused by: javax.xml.stream.XMLStreamException: Trying to output second root, <ServerCertificate> at com.ctc.wstx.sw.BaseStreamWriter.throwOutputError(BaseStreamWriter.java:1537)[78:woodstox-core-asl:4.4.1] at com.ctc.wstx.sw.BaseStreamWriter.throwOutputError(BaseStreamWriter.java:1544)[78:woodstox-core-asl:4.4.1] at com.ctc.wstx.sw.BaseStreamWriter.reportNwfStructure(BaseStreamWriter.java:1572)[78:woodstox-core-asl:4.4.1] at com.ctc.wstx.sw.BaseNsStreamWriter.checkStartElement(BaseNsStreamWriter.java:469)[78:woodstox-core-asl:4.4.1] at com.ctc.wstx.sw.BaseNsStreamWriter.writeStartElement(BaseNsStreamWriter.java:290)[78:woodstox-core-asl:4.4.1] at org.eclipse.milo.opcua.stack.core.serialization.xml.XmlEncoder.write(XmlEncoder.java:375)[73:org.eclipse.milo.stack-core:0.1.0] ... 13 more`

Thank you.

kevinherron, 2017-02-15:
This seems to be an problem in camel as it tries to toString() something
with eventually calls a method on XmlEncoder that probably doesn't work
(XmlEncoder is not fully implemented or supported right now)

ctron, 2017-02-15:
Ok so that may be my fault then, I forwarded him to milo. The original issue was ctron/de.dentrassi.camel.milo#10.

So what happens is that the `SubscriptionManager` selects a single `EndpointDescription` and then (for logging) passes this on to `EndpointDescription.encode(ep, new XmlEncoder(sw));` in order to render it as a string.

What would be the right approach then?

kevinherron, 2017-02-15:
Does `EndpointDescription` (and every other struct) not have a `toString()` in 0.1.0?

I'm not sure if `XmlEncoder` works, especially using it like that. Typically those encode/decode methods are not called directly. Rather, you put the EndpointDescription in an ExtensionObject and that gets serialized as part of some other process down the line...

ctron, 2017-02-16:
Ok. So let's close this one here. Sorry for the confusion.

closed 2017-02-16 by ctron

kevinherron, 2017-02-16:
TBH, I still don't understand the root of this: `javax.xml.stream.XMLStreamException: Trying to output second root, <ServerCertificate>`

ctron, 2017-02-17:
Looks like. But I just followed your suggest to use the native `toString()`. I am not sure why I did this, maybe there was no `toString()` method in the past, maybe I thought it was a brilliant idea back then :wink: