eclipse-milo/milo #108: Writing fields of an array

MCOberle, 2017-03-07:
Does the current version already allow to write a single field of an array?
According to [Beckhoff Infosys](https://infosys.beckhoff.de/index.php?content=../content/1031/tcopcuaserver/1865279883.html&id=) this should be possible by using the RangeOffset. However, I cannot find a method that allows setting this method.

Of course a workaround is to first read the entire array and then just change the value that needs to be changed and writing the whole array again. However, it would be nice if writing just a single field was possible as well.

kevinherron, 2017-03-07:
Yes, the `indexRange` parameter of a `WriteValue` can specify a single index or a range to write to within the array.

Please feel free to stop by the gitter.im channel or use the mailing list if you have further questions and use GitHub issues for bug reports or feature tickets in the future 👍

kevinherron, 2017-03-07:
The `indexRange` parameter is a String that is supposed to represent a `NumericRange` construct, described in Part 4 section 7.22.

The basics are that you "0", or "6" would specify index 0 or 6...
"5:7" specifies the range 5 to 7...
"1,1" specifies the [1,1] element in a two dimensional array...

closed 2017-03-07 by kevinherron