eclipse-milo/milo #115: History read interface issue

licongchao, 2017-06-05:
Finally, i moved to Milo library to implement OPCUA stuff. After some test codes, i can connect the OPCUA server with certificate files successfully, then i try to read the history data of some Node, i cannot find any Milo example for the history interface. 
My history read code is as belows:
`		
CompletableFuture<HistoryReadResponse> historyRead = client.historyRead(new ReadRawModifiedDetails(false, new DateTime(TimeUtil.transform("2016-06-02 00:00:00", TimeUtil.YYYYDMMDDD_HHCMMCSS)), 
				new DateTime(TimeUtil.transform("2016-06-03 00:00:00", TimeUtil.YYYYDMMDDD_HHCMMCSS)), uint(100), false),
				TimestampsToReturn.Both, true, historyReadIds);`

historyRead.get()

i also tried CompletableFuture but failed, Could someone help me to figure it out? BTW, i can read the history data via opcuafundation library. Thanks in advance.

licongchao, 2017-06-05:
`		CompletableFuture<HistoryReadResponse> historyRead = client.historyRead(new ReadRawModifiedDetails(false, new DateTime(TimeUtil.transform("2016-06-02 01:00:00", TimeUtil.YYYYDMMDDD_HHCMMCSS)), 
				new DateTime(TimeUtil.transform("2016-06-04 00:00:00", TimeUtil.YYYYDMMDDD_HHCMMCSS)), uint(1000), true),
				TimestampsToReturn.Both, false, historyReadIds);`

i can get the history data successfully.

closed 2017-06-05 by kevinherron

manijndl, 2018-05-23:
hi @licongchao  can u please share the whole example i will be thanful to u