eclipse-milo/milo #122: Customer type

proga2017, 2017-06-26:
Hello,

I am trying to register a customer data type as an output parameter of a method.
- I have implemented the data type class, with bin and xml encoders / decoders registration
- I have created a datatype node (using same IDs as used in the data type class)
- I have created bin / xml encoding nodes (using same IDs as used in the data type class)
- I have created description nodes for the encoding nodes

When I'm browsing my server with a DEMO client, I can see my method, but when I check the output parameters, specially the one of customer data type, I see it has the data type i=4294967295, what I think is something like "unknown"?

Can anybody help me?
Where can I find some examples how to use customer datatypes in Methods.

Thank you very much
proga

kevinherron, 2017-06-26:
I'm not 100% sure if this is even possible until version 0.2.0 is released.

That being said... it looks like you've got everything covered except having the data type dictionary present in the address space. If you use UaModeler and export to XML you can get an idea of all the different nodes required and it also generates the dictionary for you.

proga2017, 2017-06-26:
Hi Kevin,

thank you for your answer.
Is there some example how to do that, I even do not find the class UaModeler ;(

kevinherron, 2017-06-26:
UaModeler is a tool by Unified Automation: https://www.unified-automation.com/products/development-tools/uamodeler.html

It's used to model the address space. The resulting model can be exported to XML in the UANodeSet format.

There's no example for doing custom datatypes with Milo right now. You're not the only one who has asked, and I'm going to try to get a working example included with or soon after the 0.2.0 release.

proga2017, 2017-06-26:
Thank you, Kevin!

proga2017, 2017-06-27:
Kevin, I already have a nodeset XML. But thank you for the info about UaModeler, it's cool tool to see the node relations in the graphical view. I've seen that I have already created everything needed. The problematic code was in the static method AnnotationBasedInvocationHandler.fromAnnotatedObject and there the internal private method getDataType, it resolvs a node id by the given Class instance.

A good interface would be, I think, to give the method fromAnnotatedObject some optional parameter of a special interface, something like DataTypeResolver, which could be implemented by the own code.

kevinherron, 2017-06-27:
Ok I may have misunderstood your initial problem. I didn't realize you were using `AnnotationBasedInvocationHandler`. Maybe until you verify everything else works you can just use your own simple `MethodInvocationHandler` implementation.

closed 2017-06-29 by kevinherron