eclipse-milo/milo #123: Is the CompletableFuture necessary when running the client examples?

AndreGuerra123, 2017-06-27:
Hello,

I'm building a simple UI class for connecting to OPCUA servers.
The user selects between (None,  Sign and Sign and Encrypt), then between (None, B256 , B256S256), etc... which basically is selecting the are linked to SecurityPolicy.(Type). An Identity Provider ( Anonymous, User&Pass or Certificate/Private key) is also created. 

Then I build my own OpcUAClient Configurator which receives these as arguments and outputs a OpcUaClient. Mind the fact I never initialize/declared any CompletableFuture<OpcUaClient> in any step of my classes. Is it necessary? The issue I'm currently having is that I  can do client.connect(), but I cannot do client.connect.get(); (Connection refused no more information.) I'm using Prosys OPCUA Simulation Server and I'm getting endpoints (EndpointsDescriptions[]) by the stack.

 I can barely understand the ClientExample and ClientExampleRunner implementations, but I thing its not practical to build a different ClientExampleRunner class every time I want to change parameters of my connection/security access.

I drop here part of my OpcUaConfigurator, hoping you could help me out... cheers.

public class OPCUAClientConfigurator {
	private EndpointDescription endpoint;
	private IdentityProvider id;
	private OpcUaClient client;
	private int Timeout;
	private PrivateKey privatekey;
	private X509Certificate certificate;
	private PublicKey publickey;
	public OPCUAClientConfigurator(EndpointDescription endpoint, IdentityProvider id, int Timeout, X509Certificate certificate, PrivateKey privatekey, PublicKey publickey) throws Exception {
		this.endpoint = endpoint;
		this.id = id;
		this.Timeout = Timeout;
		this.certificate = certificate;
		this.privatekey = privatekey;
		this.publickey = publickey;
		this.client = createClient();
	}

	private OpcUaClient createClient() throws Exception {
		if(id instanceof UsernameProvider){
			OpcUaClientConfig config = OpcUaClientConfig.builder()
					.setApplicationName(LocalizedText.english("ABIOPAT client"))
					.setApplicationUri("urn:abiopat:client")
					.setEndpoint(endpoint)
					.setIdentityProvider(id)
					.setRequestTimeout(uint(5000))
					.build();
			return new OpcUaClient(config);
		}else if (id instanceof X509IdentityProvider){
			OpcUaClientConfig config = OpcUaClientConfig.builder()
					.setApplicationName(LocalizedText.english("ABIOPAT client"))
					.setApplicationUri("urn:abiopat:client")
					.setEndpoint(endpoint)
					.setIdentityProvider(id)
					.setCertificate(certificate)
					.setKeyPair(new KeyPair(publickey, privatekey))
					.setRequestTimeout(uint(5000))
					.build();
			return new OpcUaClient(config);
		}else{
			OpcUaClientConfig config = OpcUaClientConfig.builder()
					.setApplicationName(LocalizedText.english("ABIOPAT client"))
					.setApplicationUri("urn:abiopat:client")
					.setEndpoint(endpoint)
					.setIdentityProvider(id)
					.setRequestTimeout(uint(Timeout))
					.build();
			return new OpcUaClient(config);
		}
	}
	public OpcUaClient getOPCUAClient(){
		return client;
	}
}

kevinherron, 2017-06-27:
You don't need to use or create a ClientRunner every time, they're just part of the example.

Print out the details (endpoint URL, etc...) of the endpoint you've selected and figure out why you're getting "connection refused" - that's the real issue here. You're probably getting an endpoint URL from Prosys that you can't resolve.

AndreGuerra123, 2017-06-27:
Hello, Kevin.
When I print the selected endpoint I get this:
EndpointDescription{EndpointUrl=opc.tcp://ceamphd82.campus.ncl.ac.uk:53530/OPCUA/SimulationServer, Server=ApplicationDescription{ApplicationUri=urn:ceamphd82.campus.ncl.ac.uk:OPCUA:SimulationServer, ProductUri=urn:prosysopc.com:OPCUA:SimulationServer, ApplicationName=LocalizedText{text=SimulationServer, locale=}, ApplicationType=Server, GatewayServerUri=null, DiscoveryProfileUri=null, DiscoveryUrls=[opc.tcp://ceamphd82.campus.ncl.ac.uk:53530/OPCUA/SimulationServer]}, ServerCertificate=ByteString{bytes=[48 ... -24, 90, 84, 81]}, SecurityMode=None, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#None, UserIdentityTokens=[UserTokenPolicy{PolicyId=anonymous, TokenType=Anonymous, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=null}, UserTokenPolicy{PolicyId=external_token_basic128Rsa15, TokenType=IssuedToken, IssuedTokenType=http://docs.oasis-open.org/wss/oasis-wss-kerberos-token-profile-1.1, IssuerEndpointUrl=\\<realm>\<server principal name>, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15}, UserTokenPolicy{PolicyId=certificate_basic256, TokenType=Certificate, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#Basic256}, UserTokenPolicy{PolicyId=username_basic128, TokenType=UserName, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15}, UserTokenPolicy{PolicyId=external_token_basic256, TokenType=IssuedToken, IssuedTokenType=http://docs.oasis-open.org/wss/oasis-wss-kerberos-token-profile-1.1, IssuerEndpointUrl=\\<realm>\<server principal name>, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#Basic256}, UserTokenPolicy{PolicyId=certificate_basic128, TokenType=Certificate, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15}, UserTokenPolicy{PolicyId=username_basic256, TokenType=UserName, IssuedTokenType=null, IssuerEndpointUrl=null, SecurityPolicyUri=http://opcfoundation.org/UA/SecurityPolicy#Basic256}], TransportProfileUri=http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary, SecurityLevel=0}

I decided to change your ClientExampleRunner, so that it takes two arguments (the client ExampleInterface and the Configured Client) in the constructor:

`public class OPCUARunner {
	
	private OPCUAClientInstance clientinstance;
	private CompletableFuture<OpcUaClient> future =  new CompletableFuture<OpcUaClient>();
	private OpcUaClient client;

	public OPCUARunner(OPCUAClientInstance clientinstance, OpcUaClient client){
		this.client = client;
		this.clientinstance = clientinstance;
	}
	
	 public void run() throws Exception {
	        future.whenComplete((client, ex) -> {
	        	
	            if (client != null) {
	              
	                    try {
							client.disconnect().get();
						} catch (InterruptedException | ExecutionException e) {
							e.printStackTrace();
						}
	                    Stack.releaseSharedResources();
	               
	            } else {
	                Stack.releaseSharedResources();
	            }

	            	            
	        });

	        try {
	          
	            try {
	                clientinstance.run(client, future);
	                future.get(10, TimeUnit.SECONDS);
	            } catch (Throwable t) {
	                future.complete(client);
	                throw new Exception(t.getMessage());
	            }
	        } catch (Throwable t) {
	            future.completeExceptionally(t);
	            throw new Exception(t.getMessage());
	        }
	    }

}
`
and my ClientExample implementation to check for valid connections:

`public class OPCUATryClientInstance implements OPCUAClientInstance {

	@Override
	public void run(OpcUaClient client, CompletableFuture<OpcUaClient> future) throws Exception {
		client.connect().get();
		client.disconnect().get();
		future.complete(client);
		
	}

}`

when I do 

new OPCUARunner (new OPCUATryClientInstance, client).run(); 
Note: Client comes from OPCUAClientConfigurator showed above.
I get :-1: 

SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
java.lang.Exception: java.net.ConnectException: Connection refused: no further information: ceamphd82.campus.ncl.ac.uk/10.56.97.226:53530
	at dataacquisition.AgentsDialogs.NewProcessDataAcquisitionAB.Common.OPCUA.OPCUARunner.run(OPCUARunner.java:51)

The URL that I inserted in the UaStack retrieves the different valid endpoints from the server. I sincerely cannot figure this out.  

Cheers, André

kevinherron, 2017-06-27:
The server is returning endpoints containing URLs that you apparently can't reach or resolve.

This is unfortunately somewhat common among misconfigured servers. You'll have to manually modify the endpoints you get back from the server, replacing the endpoint URLs you receive with the one you used for discovery. See this question on SO: https://stackoverflow.com/questions/40554198/java-opc-ua-client-eclipse-milo-endpoint-url-changes-to-localhost

Or you can get that server properly configured so that it returns endpoints with URLs accessible to the outside world.

AndreGuerra123, 2017-06-28:
Ok, so it seems. I configured the Server properly, I think the issue was in the Bind Addresses. Thank you. But I still want to ask a question: What is the difference between OpcUaClient and UaClient? Cheers.

closed 2017-06-29 by kevinherron