eclipse-milo/milo #124: Can this project import OPC-UA XML?

maul5, 2017-06-27:
Hello,

I want to know if the tool generated XML can be used in "milo".

the tool is opcus-modeler(https://github.com/FreeOpcUa/opcua-modeler).

Thank you very much.

kevinherron, 2017-06-27:
I am working on [this project](https://github.com/digitalpetri/uanodeset-parser) which will make it possible, but the API is not stable yet.

I cannot include this functionality directly in Milo because of some restrictions by the Eclipse Foundation IP review team about the use of the UANodeSet and other XML files.

maul5, 2017-06-28:
Thank you for answer the letter

I will try this project. ^^

kevinherron, 2017-06-29:
There's no example yet, but hopefully soon.

Please use the mailing list or gitter.im channel for questions and save the GH issue tracker for bug reports.

closed 2017-06-29 by kevinherron