eclipse-milo/milo #127: OPC-SDK-Server does not import with MAVEN into eclipse.

Muddasir-Shakil, 2017-07-24:
Dear Sir
I am trying to learn more about eclipse milo. I am just starting with it problem that I am facing is when i import milo project in to eclipse with MAVEN I get all other packages just fine like SDK CLIENT, CORE and STACK but I can not import OPC UA SERVER and thus I can not build any example since it is being used in every example. 
I have attached the screen shots, it includes ExampleServer as well. 

[image]
[image]

ctron, 2017-07-24:
As there are many ways to import things into Eclipse, could you please describe how you did it?

The last time I did that a few days ago, using "Import …" → "Maven / Import Existing Projects", it worked fine.

Muddasir-Shakil, 2017-07-24:
I did the same way in to ecplipse using import-> Maven->Import Existing Pojects, It is working fine for OPC UA SDK Client Core and Stack but as you can see in the second snapshot I am missing OPC UA SDK server, there is nothing in the folder.

ctron, 2017-07-24:
This is how it looks for me:
[image]

Muddasir-Shakil, 2017-07-24:
this is how mine looks like :(
[image]

ctron, 2017-07-24:
So maybe you deleted some part of the code by accident. It is in the repository: https://github.com/eclipse/milo/tree/master/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server

So I would suggest to do a fresh checkout.

Muddasir-Shakil, 2017-07-24:
I just checked my local repository and this folder exsits there along with all the files from where i am creating Maven project also other file like client and Core exists too but I have problem only with it I have downloaded it twice and build the project but i got sam errors. OPC ua SDK Server still mising.

Muddasir-Shakil, 2017-07-24:
Ok I do not know the roots of the problem but i that much i know that i was using Eclipse Oxygen but when I created Maven project with Neon it works fine.

closed 2017-07-24 by kevinherron