eclipse-milo/milo #130: High amount of subscriptions / connection pooling

immzi, 2017-08-22:
Hi Kevin,

We are currently working on an application that communicates to the OPC-UA server that is a part of the WinCC OA SCADA system.
The mechanism is subscription-based, so we basically subscribe to some data points within the OPC-UA server, and once the change event fires, apply some business logic to the data and store it in a postgresql database.
The amount of simultaneous subscriptions is approximately 3000.

Currently we simply wrap the OpcUaClient instance in our wrapper class that contains some connect/disconnect/subscribe/get/set logic using the client.
Bue due to a slightly high amount of subscriptions we would like to make use of some sort of parallelization so that all 3000 subscriptions do not use a single OPC-UA connection.
The good idea that comes in mind is to implement a connection pool. If I understand the current paradigm properly, single OpcUaClient handles a single connection.
So it seems the correct way would be to implement and maintain a pool of OpcUaClient instances.

Could you maybe share any ideas/good practices/thoughts on how to do it in a correct way?
Or maybe we chose a wrong path and this approach should be avoided?

Thanks in advance.

kevinherron, 2017-08-22:
Are you really creating 3000 Subscriptions or do you have 3000 MonitoredItems? If you do have that many Subscriptions, how many items in each?

immzi, 2017-08-22:
We have approximately 3000 MonitoredItems spread within 30 Subscriptions, sorry for ambiguity.

kevinherron, 2017-08-22:
Okay. That's probably more subscriptions than is necessary for 3000 items, but if there's some logic to the way they are grouped it's probably fine.

Either way, I wouldn't consider this anywhere near a high load and you don't need to consider pooling clients. I can't imagine what it would gain you.

closed 2017-08-24 by kevinherron