eclipse-milo/milo #136: Replacement Sampling Interval min value(100ms) realted subscription

jihwanseo, 2017-09-08:
Hi.
Accroding to OPC Foundation Day Event,
At least subscription can cover 1 ms interval.
And we tested that milo sdk can also cover it on Linux (Desktop PC).
To conclude, although milo is implemented on JAVA,
It is better to provide 1 ms interval by default.

If you agree upper opinion.
Please let me know how to access it related push permission.

kevinherron, 2017-09-08:
I've made this change. If you build from source you should be able to use it, otherwise it will be included in version 1.0.4 which will be released soon.

closed 2017-09-08 by kevinherron

jihwanseo, 2017-09-11:
Thank you for your support. I'll wait for version 1.0.4. :+1:

kevinherron, 2017-09-11:
@jihwanseo version 0.1.4 is released. (It was a typo when I said 1.0.4)