eclipse-milo/milo #138: DateTime should provide conversion to Instant

ctron, 2017-09-12:
`DateTime` currently provides a conversion to Java `Date`. But it does not provide a conversion to `Instant`, which would also support nano-seconds, which `Date` does not.

kevinherron, 2017-09-20:
I have not used this new API but it seems reasonable.

The implementation will have to be smarter than just `getJavaDate().toInstant()` in order to preserve any nanosecond precision there might be.

ctron, 2017-09-20:
> The implementation will have to be smarter than just getJavaDate().toInstant() in order to preserve any nanosecond precision there might be.

That is why I raised this issue instead of a PR 😁

point85, 2017-11-07:
If millisecond precision for an application is "good enough", these two methods will be useful:

	// create a UTC ZonedDateTime from the DateTime
	public static ZonedDateTime utcTimeFromDateTime(DateTime dateTime) {
		long epochMilli = dateTime.getJavaTime();
		Instant instant = Instant.ofEpochMilli(epochMilli);
		ZonedDateTime time = ZonedDateTime.ofInstant(instant, ZoneId.of("Z"));
		return time;
	}
	
	// create a local ZonedDateTime from the DateTime
	public static ZonedDateTime localTimeFromDateTime(DateTime dateTime) {
		ZonedDateTime utc = utcTimeFromDateTime(dateTime);
		ZonedDateTime time = utc.withZoneSameInstant(ZoneId.systemDefault());
		return time;
	}

PR #382 (2018-12-21): DateTime to support Instant
Fixes #138
commit 76605d77 (2018-12-21): DateTime support conversion to Instant (#382)

fixes #138
closed 2018-12-21 by kevinherron