eclipse-milo/milo #139: EDL v1.0 license is contained in LICENSE file?

jihwanseo, 2017-10-11:
Hi. 

I have some question about License for this open source.

As you know, there are double license information (EPL v1.0 and EDL v1.0) in fisrt line of source codes.
And also the site in below introduce this open source as double licenses (EPL v1.0 and EDL v1.0)
- https://projects.eclipse.org/proposals/milo

However LICENSE file seems like there is no EDL v1.0 license. 
I have some confusion about this information.
Could you tell me detailed license information whethere EDL v1.0 is contained or not?

kevinherron, 2017-10-11:
Both licenses are included in HTML format:

https://github.com/eclipse/milo/blob/master/edl-v10.html
https://github.com/eclipse/milo/blob/master/epl-v10.html

I'll have to ask somebody whether the LICENSE file should still exist, and if so, whether it should contain both licenses or if it even matters.

kevinherron, 2017-11-22:
This will be addressed in #167.

closed 2017-11-22 by kevinherron