eclipse-milo/milo #141: BadFilterNotAllowed Result related PercentDeadband

jihwanseo, 2017-10-13:
Hi.

I am testing SDK-Server through CTT provided OPC-Fundation.
And i have some doubt about ModifyMonitoredItems

When i modify one or more monitored items specifying a filter that the server does not support, such as PercentDeadband. I get Bad_FilterNotAllowed(0x80450000) Result from SDK-Server of Milo
And, I confirmed CTT that expected result should be Bad_MonitoredItemFilterUnsupported.

To conclude, I think it is better to replace Bad_FilterNotAllowed of MonitoredDatItem.java (about Line 158) with Bad_MonitoredItemFilterUnsupported.

please let me know. if any opinion?

kevinherron, 2017-10-13:
What version of the CTT are you testing with?

If this is problem with CTT version 1.03.340.358 then of course we can change the StatusCode returned.

It's been a while since I ran the CTT against the server, so I imagine more issues may pop up.

jihwanseo, 2017-10-16:
we checked the TC from Monitored Item Services -> Monitored Items Daraband Filter -> Err-006 
and i ran CTT version 1.2.336.273 (UA Stack Version : 1.02.334.0.0) on Linux Official

closed 2018-02-03 by kevinherron