eclipse-milo/milo #142: Question: OpcUaClientConfig from xml?

marinator86, 2017-10-23:
Dear contributors,

I have [read](https://webapps.stackexchange.com/questions/53170/how-to-ask-questions-on-github) that creating an issue is a way to communicate with the authors, so I chose this way to ask a question to you. 

I'd like to know whether it is possible to create one or more instances of [OpcUaClientConfig](https://github.com/eclipse/milo/blob/master/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/api/config/OpcUaClientConfig.java) from a xml configuration file. Example here: https://github.com/OPCFoundation/UA-.NET/blob/master/ConfigurationTool/Application/Opc.Ua.ConfigurationTool.Config.xml

If no parser exists yet, can you roughly estimate the effort it takes to write one?

Thanks for your response!

Mario

kevinherron, 2017-10-23:
The XML schema used in that example appears to be specifically for use with the .NET SDK. It does not map at all to the configuration parameters in Milo's `OpcUaClientConfig`.

You could probably create your own XML representation fairly easily using one of Java's XML libraries.

marinator86, 2017-10-23:
Thanks Kevin. Just to clarify, I have seen that schema definition in the link I posted: 
`<ApplicationConfiguration xmlns:i="http://www.w3.org/2001/XMLSchema-instance" xmlns="http://opcfoundation.org/UA/SDK/Configuration.xsd">`

I thought that Configuration.xsd described a schema for "application configuration" defined by the standard. I thought that some of the parameters there map to your config classes. Example:

[UaTcpStackClientConfig](https://github.com/eclipse/milo/blob/master/opc-ua-stack/stack-client/src/main/java/org/eclipse/milo/opcua/stack/client/config/UaTcpStackClientConfig.java) contains parameters ApplicationName and ApplicationURI, which are also defined in this [xml](https://github.com/OPCFoundation/UA-.NET/blob/master/ConfigurationTool/Application/Opc.Ua.ConfigurationTool.Config.xml) I posted above.

That made me think this configuration is an official schema that OPC clients can use to get configured. Though I'm unable to find more information or the schema defintion itself.

Do you have a hint, source or other point to start?

marinator86, 2017-10-24:
ok, alright. Thanks Kevin.

closed 2017-10-24 by marinator86