eclipse-milo/milo #143: How to Contribute Milo?

jihwanseo, 2017-11-03:
Hi.

I am testing SDK through CTT provided from OPC-Fundation.
and I found out some issues to handling error code.

I'm facing some error message 'Access Deny' from git
when i try to commit our modified source code into the milo.

------------------------------------------------------------------------
ERROR: Permission to eclipse/milo.git denied to ChaJiWon.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
------------------------------------------------------------------------

I think we need to get permission to access.
could you tell me your opinion?

ctron, 2017-11-03:
Please have a look at the CONTRIBUTING file from Eclipse Hono, it is a different Eclipse project, but the same is true for Eclipse Milo: https://github.com/eclipse/hono/blob/master/CONTRIBUTING.md

kevinherron, 2017-11-03:
Yes, the CONTRIBUTING from Hono is basically the same process. 

Standard fork/PR model + make sure you have an Eclipse contributor license on file.

Milo also uses checkstyle, the configuration for which is available [here](https://github.com/eclipse/milo/blob/master/build-tools/src/main/resources/milo/checkstyle.xml). All PRs get built, tested, and checkstyled upon submission.

kevinherron, 2017-11-03:
If you're going to be fixing failed CTT tests, please help me keep it organized by creating one issue here in the tracker per failing test case, and keeping each PR with a fix confined to one issue / test case.

kevinherron, 2017-11-07:
A [CONTRIBUTING.md](https://github.com/eclipse/milo/blob/master/CONTRIBUTING.md) file now exists as well.

closed 2017-11-07 by kevinherron