eclipse-milo/milo #146: Roadmap

lln-ijinus, 2017-11-06:
Would it be possible to have a roadmap about the project ?

Not something to big, just to know which parts of the server / client /stack are done, what big features need to be done, and what's in the next version (and possibly an aproximative due date)

For example, I'm searching for historical events in the server, but I don't know if I it's implemented ?
Also, I would like to help but don't know what is needed (some directed bug fix issues could be nice, too)

Anyway, thanks a lot for the project !

kevinherron, 2017-11-06:
Yes, excellent idea.

jamesmarkchan, 2017-11-07:
Yeah a roadmap could really help organize contributions and clarify
priorities. A bulleted list under release version targets has pretty low
overhead.

kevinherron, 2017-11-22:
So I still don't have a traditional roadmap, but... as you can see by the state of the issues and PRs I'm trying to dump the TODO list from my head into GitHub in order to provide a little more transparency about what's happening now and what's on its way.

closed 2017-11-30 by kevinherron