eclipse-milo/milo #147: getBuildInfo() throws a class cast exception

point85, 2017-11-07:
When I execute the following example code to get a server BuildInfo object, I get a class cast exception:		
		// Get a typed reference to the Server object: ServerNode
        ServerNode serverNode = opcUaClient.getAddressSpace().getObjectNode(
            Identifiers.Server,
            ServerNode.class
        ).get();

        // Get a typed reference to the ServerStatus variable
        // component and read value attributes individually
        ServerStatusNode serverStatusNode = serverNode.serverStatus().get();
        BuildInfo buildInfo = serverStatusNode.getBuildInfo().get();

the call to getBuildInfo() throws this exception:		
java.util.concurrent.ExecutionException: java.lang.ClassCastException: Cannot cast org.eclipse.milo.opcua.sdk.client.model.nodes.variables.BaseDataVariableNode to org.eclipse.milo.opcua.sdk.client.model.nodes.variables.BuildInfoNode

kevinherron, 2017-11-07:
Are you testing this against the Milo example server?

point85, 2017-11-07:
It is the Unified Automation ANSI C demo server. I think the C++ one behaves the same way. A read of the product node works.

kevinherron, 2017-11-07:
This is happening because the server returns the wrong type definition for the BuildInfo node. It should return BuildInfoType but instead returns BaseDataVariableType.

You can read the value of the node without relying on the strongly typed information model like this:

```java
VariableNode buildInfoNode = client.getAddressSpace()
    .createVariableNode(Identifiers.Server_ServerStatus_BuildInfo);

BuildInfo buildInfo = buildInfoNode.getValue()
    .thenApply(ExtensionObject.class::cast)
    .thenApply(ExtensionObject::<BuildInfo>decode)
    .get();
```

point85, 2017-11-07:
Great, thanks Kevin.  It works now.

kevinherron, 2017-11-07:
I'm pretty sure I reported this as a bug to Unified Automation at the interop in September. Hopefully it's fixed in a subsequent release of their SDK and server.

closed 2017-11-07 by kevinherron