eclipse-milo/milo #152: Incorrect array dimensions returned?

point85, 2017-11-15:
When browsing an array node of the Milo example server (version 0.1.5) using the ProSys client (see attached screen capture), the array dimension is returned as [0].  Shouldn't it be [4]?
[image]

kevinherron, 2017-11-15:
No, the ArrayDimensions attribute describes the capability/capacity of each dimensions, not the current length. Zero means it's variable length.

closed 2017-11-15 by kevinherron

point85, 2017-11-15:
Thanks Kevin.  I guess the server vendor can put whatever value that they
want to in that attribute.  Attached are two screenshots from a Prosys
server with a Prosys client showing more intuitive values of [10] and [2,
3, 4].

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

kevinherron, 2017-11-15:
I suppose they can technically put whatever they want, but if you wrote an array of length 20 successfully to a node that had a value of [10] that would arguably be incorrect behavior on the server's part. Doubly so if it didn't then update the ArrayDimensions attribute to reflect the new, larger length.