eclipse-milo/milo #165: Getting TimeOutHint

ChaJiWon, 2017-11-22:
Hello, I was wonder about getTimeoutHint() in OpcUaSubscriptionManager.java(sdk-client)
I know that getTimeoutHint() is fuction that calcuating timeoutHint variable.

But I worried about excepiton case of getTimeoutHint().
If calculated timeoutHint variable is bigger than max value of interger type, it needs to check exception.

So, I think add timeoutHint validation check logic.
I suggest code like this,
        try {
            UInteger ret = uint(timeoutHint);
            return ret;
        } catch (Exception e) {
            return UInteger.MAX;
        }

What do you think about this change? (I think it needs to consider MIN case)

kevinherron, 2017-11-22:
While it's unlikely that you could have enough publish requests or a publishing interval big enough, let's entertain the idea...

Perhaps something like this implementation, which checks to see if the multiplications overflowed:

```java
    private UInteger getTimeoutHint() {
        double minKeepAlive = subscriptions.values().stream()
            .map(s -> s.getRevisedPublishingInterval() * s.getRevisedMaxKeepAliveCount().doubleValue())
            .min(Comparator.naturalOrder())
            .orElse(client.getConfig().getRequestTimeout().doubleValue());

        long maxPendingPublishes = getMaxPendingPublishes();

        double timeoutHint = maxPendingPublishes * minKeepAlive * 1.25;

        if (Double.isInfinite(timeoutHint)) {
            timeoutHint = UInteger.MAX_VALUE;
        } else if (timeoutHint > UInteger.MAX_VALUE) {
            timeoutHint = UInteger.MAX_VALUE;
        }

        logger.debug(
            "getTimeoutHint() minKeepAlive={} maxPendingPublishes={} timeoutHint={}",
            minKeepAlive, maxPendingPublishes, timeoutHint);

        return uint((long) timeoutHint);
    }
```

Now I'm noticing that this is calculated on every sent PublishRequest. Maybe it should be calculated and stored when a subscription is created/modified/deleted.

ChaJiWon, 2017-11-23:
Above code looks good. I agree that updating calculated value when a subscription is created/modified/deleted. When add new subscription time or remove subscription time, it is enough to change timeoutHint value.

closed 2017-11-26 by kevinherron