eclipse-milo/milo #166: Implement OpcUaXmlStreamEncoder

kevinherron, 2017-11-22:
If anybody is ever interested in the HTTPS transport being implemented, the implementation of `OpcUaXmlStreamEncoder` is a prerequisite.

Currently this class only exists in branch https://github.com/eclipse/milo/tree/dev/0.2.x.

The dual of this class, `OpcUaXmlStreamDecoder`, has already been implemented and works well enough to decode values from XML nodesets.

system24seven, 2025-01-13:
Looking at this on dev/1.0 I have hit an issue on the decode with `nsu=<String> ` NodeIDs. I am still working on it but the URI in the JSON encoder test breaks the XML Decoder. It may also be my code, but was the decoder class fully tested or is it also a bit of a work in process?

system24seven, 2025-01-13:
Turns out I just cannot read an OPC spec to save my own life.

system24seven, 2025-01-13:
That said, the question still stands, is the decoder tested and working in it's current state? If it is, what was it tested on?

kevinherron, 2025-01-13:
It sees use in some not yet public code for loading UANodeSet files. An old version that demonstrates the general use is at https://github.com/digitalpetri/opc-ua-nodeset-parser/blob/76ff6ce459b902a6fe2d2ab154ec8fcdf2064330/src/main/java/com/digitalpetri/opcua/nodeset/util/AttributeUtil.java#L123-L136. It basically works well enough to decode the XML-encoded values embedded inside the UANodeSet files for Variable and VariableType Nodes.

Definitely not well tested.

One of the problems I had with the XML serialization was a lack of examples and reference material. These days I think the .NET stack has a complete implementation and I think that's probably the best reference.

https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Stack/Opc.Ua.Core/Types/Encoders/XmlEncoder.cs
https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Stack/Opc.Ua.Core/Types/Encoders/XmlDecoder.cs

kevinherron, 2025-01-13:
I've been pretty lukewarm on the idea of supporting anything but the TCP transport in 1.0, and right now my plan is to remove the [transport-https](https://github.com/eclipse-milo/milo/tree/dev/1.0/opc-ua-stack/transport-https) and [transport-websocket](https://github.com/eclipse-milo/milo/tree/dev/1.0/opc-ua-stack/transport-websocket) modules and move them to some kind of "incubating" repository.

But if we can get an implementation of the XML serializers working I think supporting HTTPS/XML wouldn't be terribly difficult for the future, and the .NET stack if nothing else supports it, so there should be a decent integration testing target. Right now the transport is HTTP(S)/Binary and I never found another product out there that actually implemented that combo.

system24seven, 2025-01-14:
I have started work on this whole setup, in the process I have found an example of the XML message and also the correct bits of the spec. I am currently writing to this as far as I can, and writing tests to match this and at some point I will need to work out how to third party test this with a known good implementation.

kevinherron, 2025-04-05:
Started working on this.

kevinherron, 2025-04-09:
✅

PR #1404 (2025-04-07): Implement OpcUaXmlEncoder with XMLStreamWriter
If this works out it will fix #166.

TODO:

- [x] ~~array encoding limits~~
- [x] ~~string encoding limits~~
- [x] recursion depth check
- [x] push/pop namespace from DataTypeCodec
- [x] ExtensionObject body for other encodings

edit: encoding limit checks should probably be its own ticket and target all encoder/decoder implementations. `EncodingLimits` should be modified to include explicit limits for arrays, Strings, and ByteStrings.
commit d13bfba5 (2025-04-07): Implement OpcUaXmlEncoder (#1404)

fixes #166
closed 2025-04-09 by kevinherron