eclipse-milo/milo #179: SessionActivityListener for disconnection not called in Ubuntu

ChengYuHin99, 2017-11-28:
I am testing with client sdk 0.1.5.
 When i unplug network of opcserver, in windows onSessionInactive listener is called, but in ubuntu not called.
Looking forward your help.
Thanks.

kevinherron, 2017-11-28:
If you wait longer it should work.

Netty is at the mercy of the OS as to when it receives notification that a socket has closed. In the case where you pull the network cable, you have to wait until the Keep Alive for the TCP connection has elapsed and the OS notifies Netty (thus notifying Milo).

ChengYuHin99, 2017-11-28:
How opcclient can know connection status with opcserver immediately?  Is it impossible to make onSessionInactive  called whenever pull the network cable in linux?
Does it need to implement  method like ping in opcclient side?
Or should i decrease keep alive timeout value? If so, which parameter should to be changed?

kevinherron, 2017-11-28:
You can read the server's CurrentTime and/or State node. If it succeeds, you're connected. Periodically reading one or both of these nodes can be used as a "keep alive" mechanism by your application.

[This Stack Overflow](https://stackoverflow.com/questions/1480236/does-a-tcp-socket-connection-have-a-keep-alive) thread has some information about out TCP keep alive works and how you can configure different values on Linux.  

Generally I think trying to get TCP Keep Alive to happen faster is bound to be trouble. It's best to layer your own mechanism on top if you need to determine that a connection is broken under some circumstances.

ChengYuHin99, 2017-11-28:
Ok, thanks. I will close issue.

closed 2017-11-28 by ChengYuHin99