eclipse-milo/milo #181: I want to Keep start for Subscription mode on client,How to automatic reconnect when disconnected

zk230, 2017-11-28:
I want to Keep start for Subscription mode on client,How to automatic reconnect when disconnected @kevinherron

kevinherron, 2017-11-28:
Once a connection has been established, re-connection is automatic. You don't need to enable it in anyway. The client will continue to reconnect and keep an active session until you tell it to disconnect.

zk230, 2017-11-29:
but, the  SubscriptionExample is do not reconnect,why? @kevinherron

kevinherron, 2017-11-29:
Sorry, I'm not 100% what you're asking.

All of the client examples shut themselves down after 10-15 seconds, so you won't be able to observe any reconnect behavior in those.

zk230, 2017-11-29:
I am change Thread.sleep(5000) to Thread.sleep(99999999) , the SubscriptionExample is keep long time,I am stop the server ,but SubscriptionExample is do not reconnect @kevinherron

kevinherron, 2017-11-29:
If you're looking to see more data arrive then this example won't show you that. It doesn't add a `UaSubscription.NotificationListener` that handles the transfer failure that will occur when it reconnects to the server you restarted.

See `OpcUaClientIT#testFailedSubscriptionTransfer` for a test that verifies the reconnect happens and subscriptions can resume if the listener is in place.

zk230, 2017-11-30:
Sorry, this method did not solve my problem. But I guess the location of the problem.

[ua-netty-event-loop-1] ERROR org.eclipse.milo.opcua.stack.client.handlers.UaTcpClientMessageHandler - [remote=localhost/127.0.0.1:12686] Exception caught: An existing connection was forcibly closed by the remote host.

"UaTcpClientMessageHandler"  this class is netty client,but this class no implement "channelUnregistered" method,you can refer to [https://github.com/netty/netty/blob/4.1/example/src/main/java/io/netty/example/uptime/UptimeClientHandler.java](url)
and 
[https://github.com/netty/netty/blob/4.1/example/src/main/java/io/netty/example/uptime/UptimeClient.java](url)

kevinherron, 2017-11-30:
Implementing `channelUnregistered` is not needed. `ClientChannelManager` adds an adapter that implements `channelInactive` and starts the reconnect, seen [here](https://github.com/eclipse/milo/blob/cd0da80b171d78d362cde2305214c07ad96c8478/opc-ua-stack/stack-client/src/main/java/org/eclipse/milo/opcua/stack/client/ClientChannelManager.java#L372-L390).

Can you post a code example that explains and shows the problem you're having?

kevinherron, 2017-11-30:
Also, are you aware that `ClientExampleRunner` terminates after 15 seconds regardless of the amount of time your example sleeps for: https://github.com/eclipse/milo/blob/f388968c8fcfe73da2957028a0d2c11e91d6e78c/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/ClientExampleRunner.java#L117-L123

and upon termination it disconnects the client and shuts down the server.

zk230, 2017-12-08:
I know this time limit,but when extension of this time and split client and server then stop the server, It is do not reconnect

kevinherron, 2017-12-08:
Please provide a self-contained reproducible example code that demonstrates the issue. 

All of the tests as well as production deployments of the client reconnect, so either there's some kind of communication problem between you and I or you've stumbled upon a legitimate bug that nobody else has reproduced so far.

closed 2017-12-20 by kevinherron

zk230, 2018-01-02:
hi

zk230, 2018-01-02:
@kevinherron  Hi,I'm sorry I can not reply in time,I will now explain this issue,when the client first started, the server does not start, this time can not be reconnected, and I would like to ask How to always reconnect?

zk230, 2018-01-23:
@kevinherron hello?Can you help me？

kevinherron, 2018-01-23:
The state machine only keeps reconnecting once a successful connection has been established. The user of the client SDK is responsible for calling `connect()` and ensuring it completes successfully, as well as any subsequent calls to keep trying to make the initial connection.

A very simplified example of what this could look like:

```java
    private static void connect(OpcUaClient client) {
        client.connect().whenComplete((c, ex) -> {
            if (ex != null) {
                Stack.sharedScheduledExecutor().schedule(
                    () -> connect(client), 1, TimeUnit.SECONDS);
            }
        });
    }
```

What works for your application will be different. You'll probably want to track the scheduled task, log failures, be able to cancel the task, etc...

maicb, 2019-01-11:
> 如果您希望看到更多数据到达，那么此示例将不会向您显示。它不会添加`UaSubscriptionManager.SubscriptionListener`用于处理重新连接到重新启动的服务器时发生的传输失败的情况。
> 
> 查看`OpcUaClientIT#testFailedSubscriptionTransfer`验证重新连接的测试，如果侦听器就位，则可以继续订阅。

when i restart server the sdk respone message:" Received unmatched TransferSubscriptionsResponse with requestHandle=0, timestamp=DateTime{utcTime=0, javaDate=Mon Jan 01 08:00:00 CST 1601}", can you help me out ?

0DennisLee0, 2020-02-24:
You can easily change the 
```java

            try {
                clientExample.run(client, future);
                future.get(15, TimeUnit.SECONDS);
            } catch (Throwable t) {
                log.error("Error running com.dennis.client_milo example: {}", t.getMessage(), t);
                future.completeExceptionally(t);
            }
```
to 
```java

            try {
                clientExample.run(client, future);
//                future.get(15, TimeUnit.SECONDS);
                future.get();
            } catch (Throwable t) {
                log.error("Error running com.dennis.client_milo example: {}", t.getMessage(), t);
                future.completeExceptionally(t);
            }
```
And the subscription will be carrying on