eclipse-milo/milo #189: How to implement Client Redundancy

zk230, 2017-12-08:
I have many client for subscription one opc ua server,but the server is not "milo" server,how to implement Client Redundancy

kevinherron, 2017-12-08:
Client Redundancy as described by part 4 of the spec would require that one `OpcUaClient` instance be able to transfer subscriptions to/from another instance, which isn't currently possible. 

The transfer service is callable, but there's no way for the subscription manager in the client taking over the subscriptions to actually take an `OpcUaSubscription` object and start managing it.

I'll make a ticket to implement this as an enhancement eventually, but the only way I can see it working is if the server also has diagnostics enabled so that the instance taking over subscriptions can learn the details of the subscription (publishing interval, priority, keep-alive count, lifetime count, etc...).

closed 2017-12-20 by kevinherron

marcus-orchard, 2019-08-06:
@kevinherron - would you be ok if I took a stab at implementing this as a PR and associating the commit with this enhancement request? 

I agree, the server would need to have diagnostics enabled, otherwise the client should return an error. 

Looking at the code, I think this enhancement can be achieved by adding logic to the _UaSubscriptionManager_/_OpcUaSubscriptionManager_ for transferring a subscription. This could:

1. Read the value of _ServerDiagnostics_/_EnabledFlag_ and return an error if false.
2. Call the _transferSubscriptions_ method in the client
3. Read the value of relevant _SubscriptionDiagnostics_ for the given subscription ids
4. Call _Server_/_GetMonitoredItems_ for each subscription id. This may be possible in parallel to the previous step.
5. Create an _OpcUaSubscription_ on the client for each subscription using the read diagnostics and add them to the _subscriptions_ Map in the subscription manager.
6. Using the results of the _GetMonitoredItems_ create the _OpcUaMonitoredItem_ for each subscription locally in the client. I still need to determine how I could provide all the fields for this object's definition, eg _ReadValueId_. One initial thought would be to set a _wasTransferred_ flag on each monitored item and if true when handling subscription notifications set the missing data from the information returned in the monitored item's notification. Other potential approaches could include _ResendData_ or _Republish_.
7. Setting the event/value consumer is a little tricky. I'll not be able to ascertain from the _GetMonitoredItems_ if the monitored item is a value or event. It may be necessary to set both for all transferred items, and then leverage the previously mentioned _wasTransferred_ flag to remove the extraneous consumer at runtime. 

Of course I'd be completely open to any suggestions, thoughts, collaborations, or other points of interest you could share. 

Is this something you'd be open to?

kevinherron, 2019-08-06:
@marcus-orchard I moved this into #523 for discussion.