eclipse-milo/milo #195: Error handling of QualifiedName.java (About CTT case of TranslateBrowsePathsToNodeIds)

jihwanseo, 2018-01-02:
Hi.

As you know, There is some precondition logic in below 

=======================================================
[constructure of QualifiedName.java - line 53]
...
Precondition.checkArgument(name == null || name.length() <= 512, "name");
...
=======================================================

It seems like there is no expect result from server as this precondition logic is ran.
Please refer detailed expection of CTT following below. 

- Case ID : Err-014 (TranslateBrowsePath)
- Category TranslateBrowsePathsToNodeIds() 
- Description : A relativePath element specifies an invalid targetName (i.e., too long (>512 chr$) or invalid characters (0x00 &ndash; 0x08, 0x0E &ndash; 0x1F, or 0x7F) in the name string).
- Expect Result : ServiceResult = Good.
Operation error in this case is permitted to be one of the following result codes - both are considered correct:
* Bad_BrowseNameInvalid
* Bad_NoMatch.

kevinherron, 2018-01-02:
I'll look into this a little more.

Since the spec limits the name portion to <= 512 in part 3 (i.e. it doesn't have anything to do with the TranslateBrowsePaths service) I think the CTT should allow it to fail as a decoding error in the serialization layer, which is probably where you're seeing it fail.

kevinherron, 2018-01-23:
This issue is now being tracked in the OPCF Mantis instance. The test case will eventually be revised (or removed entirely...). No change to Milo is necessary at this time.

closed 2018-01-23 by kevinherron